lnis-uofu/OpenFPGA #411: Functionality to compile only OpenFPGA engine.

aram-rs, 2021-11-04:
**Is your feature request related to a problem? Please describe.**
The feature will reduce compilation time when the user is not interested in yosys build.

**Describe the solution you'd like**
Add `compile_openfpga` target in the Makefile which will execute CMAKE with flags that will disable yosys build.

**Describe alternatives you've considered**
Alternatively a new target can be added to the Makefile which will combine checkout with the build of only OpenFPGA engine.

**Additional context**
Users having custom Synthesis tools will benefit from this future.

tangxifan, 2021-11-08:
@tpagarani @coolbreeze413 Would you mind modifying our Makefile to support this feature? It will help us in testing different yosys versions from containers. Let me know what you think.

tangxifan, 2021-12-17:
Close as PR #426 is merged

closed 2021-12-17 by tangxifan