lnis-uofu/OpenFPGA #415: Pdk integration

msaideroglu, 2021-11-09:
How can I integrate my custom pdk to OpenFpga? To do this what things should be done?

ganeshgore, 2021-11-24:
Hello @msaideroglu, to integrate our custom PDK/ Standard cell library you will have to update the OpenFPGA architecture file.
for example `name="sky130_fd_sc_hd__inv_1" ` in following line should be replaced with your own functionally equivalent  standard cell name.You need to do such mapping for the entire OpenFPGA architecture file, in order to get a correct tech mapped netlist.
https://github.com/lnis-uofu/OpenFPGA/blob/1fd9367ae9f7518fd041597e18c0650f1188dfee/openfpga_flow/openfpga_arch/k4_frac_N8_register_scan_chain_caravel_io_skywater130nm_fdhd_cc_openfpga.xml#L34

closed 2021-12-17 by tangxifan

Gurusatwik, 2024-04-11:
> Hello @msaideroglu, to integrate our custom PDK/ Standard cell library you will have to update the OpenFPGA architecture file. for example `name="sky130_fd_sc_hd__inv_1" ` in following line should [...]

Is it not compulsory to mention verilog_netlist= "</Path to .v file of PDK>" in circuit_model?