lnis-uofu/OpenFPGA #423: An Error while creating *.act file from *.blif file by using ACE

emincetin452, 2021-11-24:
Hi there,
I am trying to generate act and blif files of ${OPENFPGA_PATH}/openfpga_flow/benchmarks/micro_benchmark/counters/counter_8bit_async_resetb/counter.v file. I have created counter.blif by using yosys and here is the blif file.

```
# Generated by Yosys 0.10.0 (git sha1 dca8fb54a, gcc 9.3.0-17ubuntu1~20.04 -fPIC -Os)

.model counter
.inputs clk resetb
.outputs result[0] result[1] result[2] result[3] result[4] result[5] result[6] result[7]
.names $false
.names $true
1
.names $undef
.names result[0] $auto$alumacc.cc:485:replace_alu$5.X[0]
0 1
.names result[4] result[5] $abc$201$new_n18_
11 1
.names result[2] result[3] $abc$201$new_n19_
11 1
.names result[1] result[0] $abc$201$new_n20_
0- 1
-0 1
.names $abc$201$new_n19_ $abc$201$new_n20_ $abc$201$new_n21_
10 1
.names $abc$201$new_n21_ $abc$201$new_n18_ $abc$201$new_n22_
0- 1
-0 1
.names $abc$201$new_n22_ result[6] $auto$alumacc.cc:485:replace_alu$5.Y[6]
11 1
00 1
.names result[6] $abc$201$new_n22_ $abc$201$new_n24_
10 1
.names $abc$201$new_n24_ result[7] $auto$alumacc.cc:485:replace_alu$5.Y[7]
10 1
01 1
.names result[1] result[0] $auto$alumacc.cc:485:replace_alu$5.Y[1]
10 1
01 1
.names $abc$201$new_n20_ result[2] $auto$alumacc.cc:485:replace_alu$5.Y[2]
11 1
00 1
.names result[2] $abc$201$new_n20_ $abc$201$new_n28_
10 1
.names $abc$201$new_n28_ result[3] $auto$alumacc.cc:485:replace_alu$5.Y[3]
10 1
01 1
.names $abc$201$new_n21_ result[4] $auto$alumacc.cc:485:replace_alu$5.Y[4]
10 1
01 1
.names $abc$201$new_n21_ result[4] $abc$201$new_n31_
11 1
.names $abc$201$new_n31_ result[5] $auto$alumacc.cc:485:replace_alu$5.Y[5]
10 1
01 1
.subckt $_DFF_PN0_ C=clk D=$auto$alumacc.cc:485:replace_alu$5.X[0] Q=result[0] R=resetb
.subckt $_DFF_PN0_ C=clk D=$auto$alumacc.cc:485:replace_alu$5.Y[1] Q=result[1] R=resetb
.subckt $_DFF_PN0_ C=clk D=$auto$alumacc.cc:485:replace_alu$5.Y[2] Q=result[2] R=resetb
.subckt $_DFF_PN0_ C=clk D=$auto$alumacc.cc:485:replace_alu$5.Y[3] Q=result[3] R=resetb
.subckt $_DFF_PN0_ C=clk D=$auto$alumacc.cc:485:replace_alu$5.Y[4] Q=result[4] R=resetb
.subckt $_DFF_PN0_ C=clk D=$auto$alumacc.cc:485:replace_alu$5.Y[5] Q=result[5] R=resetb
.subckt $_DFF_PN0_ C=clk D=$auto$alumacc.cc:485:replace_alu$5.Y[6] Q=result[6] R=resetb
.subckt $_DFF_PN0_ C=clk D=$auto$alumacc.cc:485:replace_alu$5.Y[7] Q=result[7] R=resetb
.names result[1] $auto$alumacc.cc:485:replace_alu$5.X[1]
1 1
.names result[2] $auto$alumacc.cc:485:replace_alu$5.X[2]
1 1
.names result[3] $auto$alumacc.cc:485:replace_alu$5.X[3]
1 1
.names result[4] $auto$alumacc.cc:485:replace_alu$5.X[4]
1 1
.names result[5] $auto$alumacc.cc:485:replace_alu$5.X[5]
1 1
.names result[6] $auto$alumacc.cc:485:replace_alu$5.X[6]
1 1
.names result[7] $auto$alumacc.cc:485:replace_alu$5.X[7]
1 1
.names $auto$alumacc.cc:485:replace_alu$5.X[0] $auto$alumacc.cc:485:replace_alu$5.Y[0]
1 1
.end
```

When I try to generate counter.act from that counter.blif file I got that error in terminal:

```
Line 55: Cannot find the model for subcircuit $_DFF_PN0_.
Reading network from file has failed.
${OPENFPGA_PATH}/ace2/SRC/ace.c:357 main: Assertion 'ntk' failed.
Aborted (core dumped)
```

Also I have seen [that](https://github.com/lnis-uofu/OpenFPGA/issues/318#issue-906409580) discussion but It didn't help me.
Thanks

tangxifan, 2021-11-24:
@emincetin452 It means that Yosys failed to map FFs to a desired model (``.latch`` for VPR). You may refer to our yosys template scripts https://github.com/lnis-uofu/OpenFPGA/tree/master/openfpga_flow/misc

emincetin452, 2021-11-26:
> @emincetin452 It means that Yosys failed to map FFs to a desired model (`.latch` for VPR). You may refer to our yosys template scripts https://github.com/lnis-uofu/OpenFPGA/tree/master/openfpga_flow/misc

I have managed the problem with ys_tmpl_yosys_vpr_flow.ys script. Thanks a lot.

closed 2021-11-26 by emincetin452