lnis-uofu/OpenFPGA #428: Generated Testbench Fails

emincetin452, 2021-11-30:
I get errors after running Full Testbench. As you can see the below picture, after configuration phase the input ports seems like OK in GPIO pad but the output ports are settled to unknown. So the testbench failed. I have used `${OPENFPGA_ROOT}/openfpga_flow/benchmarks/micro_benchmark/counters/counter_8bit_async_resetb/counter.v` as a benchmark circuit. All the fpga is synthesized without error. So what can cause to this? 

[image]

tangxifan, 2021-12-10:
This is a known problem and should be fixed in a PR #403
It is due to that the ``greset`` is in wrong a polarity
Would you mind giving a try?

emincetin452, 2021-12-15:
Hi @tangxifan, I have changed
`assign reset[0] = greset[0];`
line with this
`assign reset[0] = ~greset[0];` as you metioned.
unknown output problem has solved. however, while benchmark circuit were properly counting, FPGA outputs still stays all zero. 
You can see:
[image]

tangxifan, 2021-12-17:
It seems a valid bug. I will look into it. Thanks a lot.

emincetin452, 2021-12-20:
One more thing about testbenches.
The default generated testbench's programmer has also problem. As you can see in the picture, `ccff_head` signal is always `X` value. 
[image]

I always change these lines 
```
always @(negedge prog_clock_reg[0]) begin
	if (bit_index >= `BITSTREAM_LENGTH) begin
		config_done[0] <= 1'b1;
	end else begin
		ccff_head[0] <= bit_mem[bit_index];
		bit_index <= bit_index + 1;
	end
end
```
with these lines.
```
always @(negedge prog_clock_reg[0]) begin
	if (bit_index >= `BITSTREAM_LENGTH) begin
		config_done[0] <= 1'b1;
	end else if(bit_index >= 0 && bit_index < `BITSTREAM_LENGTH) begin
		ccff_head[0] <= bit_mem[bit_index];
		bit_index <= bit_index + 1;
	end
end
```

tangxifan, 2021-12-20:
@emincetin452 Can you provide more details on the test case you tried?

- OpenFPGA version / commit id
- OpenFPGA test case (Any modification did on the test case?)
- HDL simulator

I have tried the full testbench with iVerilog, it works fine (see attached screenshot) 
[image]

emincetin452, 2021-12-21:
OpenFPGA version : 'origin/master'

I use this script to generate verilog design and testbench files:
[script.zip](https://github.com/lnis-uofu/OpenFPGA/files/7750101/script.zip)

I use ModelSim - INTEL FPGA STARTER EDITION 2020.1

and there is no any modification.

emincetin452, 2022-01-18:
Is there any update about that bug?

tangxifan, 2022-01-18:
@emincetin452 Regarding the additional lines in Verilog testbenches, I am o.k. to add them. If iVerilog is cool with them, then we can update the testbench generator. 
Would you mind giving a try? I can show you which file and which line to modify, if you are interested.

Regarding the reset bugs, give me some time. It will take a few more days.

emincetin452, 2022-01-27:
@tangxifan it sounds great. I want to try, let me do it.

tangxifan, 2022-01-27:
@emincetin452 No problem. Please the line and related functions:

https://github.com/lnis-uofu/OpenFPGA/blob/7ed92b732f370581789fc514d5b76fc39ea173d9/openfpga/src/fpga_verilog/verilog_top_testbench.cpp#L1434

tangxifan, 2022-02-15:
@eemincetin The issue should be resolved once PR #529  is merged.

closed 2022-02-15 by tangxifan