lnis-uofu/OpenFPGA #442: Automatic Version Number Bump

tangxifan, 2021-12-14:
**Is your feature request related to a problem? Please describe.**
Now, OpenFPGA's version number is hard coded in the CMakelist.txt and C++ source files

https://github.com/lnis-uofu/OpenFPGA/blob/08f6b9637db1e88a017ff49d42e0f5bb0850d0f2/CMakeLists.txt#L53-L57

https://github.com/lnis-uofu/OpenFPGA/blob/08f6b9637db1e88a017ff49d42e0f5bb0850d0f2/libopenfpga/libopenfpgautil/src/openfpga_version.cpp.in#L1-L20

Typically, developers should control the major and minor version number by bumping them manually.
However, the patch number can be automatically bumped up. A typical way is to bump it with commits or PRs. In the other word, the patch number represents the number of commits/PRs merged after a major&minor version released.

**Describe the solution you'd like**
YosysHQ seems to have a reasonable solution by using github bot. We can adapt it and give a try

[image]

tangxifan, 2021-12-14:
@nadeemyaseen-rs As we discussed, please look into it. Thanks!

nadeemyaseen-rs, 2021-12-17:
@tangxifan  After looking at version bump workflow file of YosysHQ, this is what I have understood:

* They do `git tag` manually when they want to do a release. Currently the tag number is 0.12
* Now in `version bump yml` file, they use this command ` git log --oneline 2156e20.. | wc -l` to count total number of commits hashes after `git tag` command.            (`2156e20..` is actually the commit hash of `git tag` command used as base for counting the number of patches) 
* In this way `yosys` version number become 0.12+36  where 36 is the total number of commits done after 0.12 version release

Now for OpenFPGA:

For major version, my understanding is that it is hard coded to 1 (correct me if I am wrong). Similarly the minor and patch number are hard coded to 0. The revision number is the last commit.    
Now to follow YosysHQ way for the patch number, what should I use as base commit to count the number of patches. In other words what commit hash should I use in `git log --oneline ???` ?

tangxifan, 2021-12-17:
@nadeemyaseen-rs Please consider the latest release as the commit hash: https://github.com/lnis-uofu/OpenFPGA/releases/tag/v0.3-alpha

Basically, your understanding on version number is correct
- The major and minor number should be hard coded. They are manually bumped by maintainers. 
- The patch number is the result of ``wc -l``
- Revision number is the last commit.

tangxifan, 2021-12-21:
@nadeemyaseen-rs Just see errors in github action

[image]

Is there a way to raise the PR, then ask us to approve and merge?

nadeemyaseen-rs, 2021-12-21:
@tangxifan  As `master` branch is protected so PR seems only the solution. Let me modify the workflow file to create an automatic PR.

tangxifan, 2021-12-21:
@nadeemyaseen-rs Thanks!

closed 2021-12-30 by tangxifan