lnis-uofu/OpenFPGA #444: How to modify the  dsp/multipler's  .v file in the openfpga's XML file

Mistal1999, 2021-12-15:
I have go into the file 'k4_frac_N8_reset_softadder_register_scan_chain_frac_dsp8_caravel_io_skywater130nm_fdhd_cc_openfpga.xml'
and replaced the .v file of dsp with my own design. And I checked the output result, such as critical path and area. But the result remained the same with previous run.  So is it the correct way to change .v file or something I missed during the modification.

tangxifan, 2021-12-15:
.v file does not contain any timing and area information. That is why you see the results are not changed.
Only when you run physical design, you can see the timing and area difference by changing the .v file.
The timing values and area values are maintained in VPR architecture .xml. If you change the numbers there, you will see a change.

Mistal1999, 2021-12-16:
Thanks, I understeand now. But I still have a question. Since the .v file in openfpga arch file can not change the result, and I implement different circuits on the same architecture(recently I have been using k4_N4_40nm_cc_openfpga.xml ), such as different approximate multilers.  Does the VPR output result can be considered as a index to evaluate the performance of these circuits

tangxifan, 2021-12-17:
@Mistal1999 The VPR output results are based on the timing values and area values in your VPR architecture files. For most VPR architecture files in OpenFPGA projects, they are designed to validate a specific technical feature of OpenFPGA tools. Currently, the focus of these test cases are on functional validation. Only the SOFA architectures are calibrated based on post layout results, which can be considered as an index to evaluate performance.

closed 2021-12-17 by tangxifan