lnis-uofu/OpenFPGA #461: Some questions about FPGA-Bitstream

kangliyu1, 2021-12-24:
Hello! When I ran the modified task/benchmark_sweep/mac_units example of OpenFPGA, and changed the benchmark to a FIFO module written by myself or another module written by myself (the FIFO code is as follows), the following errors (such as log: 1), after my own positioning, it is found that one of the two always blocks in picture 1 and one of the two always blocks in picture two can be reserved to generate bitstream, and the complete FIFO module code can be on Vivado The bitstream was successfully generated. This problem has troubled me for a long time. I hope someone can give me some advice.
```
fifo.v
module  fifo(clk,rstp,din,writep,readp,dout,emptyp,fullp
);
input           clk;
input           rstp;      // 复位信号
input[15:0]     din;
input           readp;
input           writep;
output[15:0]    dout;
output          emptyp;
output          fullp;
parameter       DEPTH = 2,
                MAX_COUNT=2'b11;      //定义地址最大值            
reg             emptyp;
reg             fullp;
reg[15:0]       dout;
reg[(DEPTH-1):0]    tail;
reg[(DEPTH-1):0]    head;
reg[(DEPTH-1):0]    count;
reg[15:0]           fifomem[0:MAX_COUNT];       //  定义fifo存储器，4个16位的存储器
// dout被赋给tail指针指向的数值
always @(posedge clk)  begin
  if(rstp==1) begin
    dout <= 16'h0000;       // 复位信号有效置0
  end
  else begin
    dout <= fifomem[tail];   //将fifomem中第tail个单元给dout
  end
end
// 写入数据
always @(posedge clk) begin
    if(rstp==1'b0 && writep == 1'b1 && fullp == 1'b0) begin
        fifomem[head]<=din;      // 写入
    end
end
// head指针递增
always @(posedge clk) begin
    if(rstp==1'b1) begin
        head<=2'b00;
    end
    else begin
        if(writep==1'b1 && fullp==1'b0) begin
        head<=head+1;
        end
    end
end
//tail指针递增
always @(posedge clk) begin
    if(rstp==1'b1) begin
        tail<=2'b00;
    end
    else begin
        if(readp==1'b1 && emptyp==1'b0) begin
        tail<=tail+1;
        end
    end
end
// 计数器
always @(posedge clk) begin
   if (rstp == 1'b1) begin
      count <= 2'b00;
   end
   else begin
      case ({readp, writep})
         2'b00: count <= count;
         2'b01: 
            if (count != MAX_COUNT) 
               count <= count + 1;     //为写状态时计数器进行加法计数
         2'b10: 
            if (count != 2'b00)
               count <= count - 1;    //为读状态计数器进行减法计数
         2'b11:
           count <= count;
      endcase
   end
end
// empty指针
always @(count) begin
   if (count == 2'b00)
      emptyp <= 1'b1;      //count为0时emptyp赋为1
   else
      emptyp <= 1'b0;
end

// fullp指针
always @(count) begin
   if (count == MAX_COUNT)
      fullp <= 1'b1;                     //计数到最大时fullp赋为1
   else
      fullp <= 1'b0;
end
endmodule
```
log:
[image]

fig1:① and ② retain only one
[image]
fig2:① and ② retain only one
[image]

您好！当我运行OpenFPGA的修改后的task/benchmark_sweep/mac_units例子时，将benchmark改成自己编写的FIFO模块或者是其他的自己写的模块时(FIFO代码如下)，都出现了下方的错（如log:1所示），经过自己的定位发现在图片1中的两个always块保留一个以及在图片二中的两部分always块保留其中一部分则可以生成bitstream，而完整的的FIFO模块代码可以在Vivado上成功生成bitstream，在OpenFPGA平台则不行，这是我写的verilog不符合OpenFPGA平台所能接受的标准吗？这个问题困扰了我许久，希望有前辈可以指点一下。

tangxifan, 2021-12-25:
Can you specify which architecture file you are using?
It seems that some nets are not routable in the architecture.

kangliyu1, 2021-12-25:
> Can you specify which architecture file you are using? It seems that some nets are not routable in the architecture.

@tangxifan Hello I am using the architecture file iwls_benchmark_example_script.openfpga

closed 2021-12-31 by kangliyu1