lnis-uofu/OpenFPGA #464: Mapping into Hard Logic

emincetin452, 2021-12-29:
Hi there. I am trying to embed a hard logic into the architecture. So far, I have added [FIFO](https://github.com/lnis-uofu/OpenFPGA/blob/master/openfpga_flow/benchmarks/micro_benchmark/fifo/rtl/fifo.v) to the given [vpr](https://github.com/lnis-uofu/OpenFPGA/blob/master/openfpga_flow/vpr_arch/k6_frac_N10_tileable_adder_chain_40nm.xml) and [openfpga](https://github.com/lnis-uofu/OpenFPGA/blob/master/openfpga_flow/openfpga_arch/k6_frac_N10_adder_chain_40nm_openfpga.xml).

Naturally, my expectation is that If I have embed that FIFO hard logic into one of my FPGA tiles, When I try to run full openfpga flow with exactly same benchmark circuit which is [FIFO](https://github.com/lnis-uofu/OpenFPGA/blob/master/openfpga_flow/benchmarks/micro_benchmark/fifo/rtl/fifo.v) It should map this benchmark circuit into the FIFO Tile. But the result is that given benchmark circuit is mapped to all the other CLBs.

What should be considered while using hard logics and mapping a circuit into it? Thanks.

tangxifan, 2021-12-29:
@emincetin452 If you want FIFO to be directly mapped, you need to 
- Ensure Yosys considers the FIFO as a blackbox. You may have to write your tech lib for Yosys.
- When Yosys can output the netlists you want, VPR should map the FIFO to hardlogic directly.

emincetin452, 2021-12-31:
@tangxifan Do you have any custom tech lib generator script example?

tangxifan, 2021-12-31:
@emincetin452 Currently, I do not. I wrote the tech lib files myself. But I think it should be a future work.

closed 2022-01-04 by emincetin452