lnis-uofu/OpenFPGA #470: Fail to execute openfpga flow

Sun1927, 2022-01-03:
Hi,  
I am trying to generate bitstream for our own fpga architecture, the result fail like this:
INFO (     MainThread) - Setting loggger in debug mode
 INFO (     MainThread) - Set up to run 2 Parallel threads
 INFO (     MainThread) - Currently running task fpga_bitstream/generate_bitstream/our
 INFO (     MainThread) - Created "run008" directory for current task run
 INFO (     MainThread) - Running "yosys_vpr" flow
 INFO (     MainThread) - Found 1 Architectures 1 Benchmarks & 1 Script Parameters
 INFO (     MainThread) - Created total 1 jobs
DEBUG (00_and2_MIN_ROUTE_CHAN_WIDTH) - Running OpenFPGA flow with [['python3', '/home/sun/OpenFPGA/openfpga_flow/scripts/run_fpga_flow.py', '/home/sun/OpenFPGA/ours/our_VTR_arch.xml', '/home/sun/OpenFPGA/pg_sample.v', '--top_module', 'and2', '--run_dir', '/home/sun/OpenFPGA/openfpga_flow/tasks/fpga_bitstream/generate_bitstream/our/run008/our_VTR_arch/and2/MIN_ROUTE_CHAN_WIDTH', '--fpga_flow', 'yosys_vpr', '--openfpga_shell_template', '/home/sun/OpenFPGA/openfpga_flow/openfpga_shell_scripts/bitstream_setting_example_script.openfpga', '--openfpga_arch_file', '/home/sun/OpenFPGA/ours/our_open_arch.xml', '--openfpga_sim_setting_file', '/home/sun/OpenFPGA/openfpga_flow/openfpga_simulation_settings/fixed_sim_openfpga.xml', '--vpr_fpga_verilog', '--vpr_fpga_verilog_dir', '/home/sun/OpenFPGA/openfpga_flow/tasks/fpga_bitstream/generate_bitstream/our/run008/our_VTR_arch/and2/MIN_ROUTE_CHAN_WIDTH', '--vpr_fpga_x2p_rename_illegal_port', '--debug', '--TOP', 'and2']]
 INFO (00_and2_MIN_ROUTE_CHAN_WIDTH) - Validating command line arguments
 INFO (00_and2_MIN_ROUTE_CHAN_WIDTH) - Setting loggger in debug mode
 INFO (00_and2_MIN_ROUTE_CHAN_WIDTH) - Run directory : /home/sun/OpenFPGA/openfpga_flow/tasks/fpga_bitstream/generate_bitstream/our/run008/our_VTR_arch/and2/MIN_ROUTE_CHAN_WIDTH
 INFO (00_and2_MIN_ROUTE_CHAN_WIDTH) - Running "yosys_vpr" Flow
 INFO (00_and2_MIN_ROUTE_CHAN_WIDTH) - Traceback (most recent call last):
 INFO (00_and2_MIN_ROUTE_CHAN_WIDTH) -   File "/home/sun/OpenFPGA/openfpga_flow/scripts/run_fpga_flow.py", line 890, in <module>
 INFO (00_and2_MIN_ROUTE_CHAN_WIDTH) -     main()
 INFO (00_and2_MIN_ROUTE_CHAN_WIDTH) -   File "/home/sun/OpenFPGA/openfpga_flow/scripts/run_fpga_flow.py", line 262, in main
 INFO (00_and2_MIN_ROUTE_CHAN_WIDTH) -     run_yosys_with_abc()
 INFO (00_and2_MIN_ROUTE_CHAN_WIDTH) -   File "/home/sun/OpenFPGA/openfpga_flow/scripts/run_fpga_flow.py", line 583, in run_yosys_with_abc
 INFO (00_and2_MIN_ROUTE_CHAN_WIDTH) -     ys_params = create_yosys_params()
 INFO (00_and2_MIN_ROUTE_CHAN_WIDTH) -   File "/home/sun/OpenFPGA/openfpga_flow/scripts/run_fpga_flow.py", line 471, in create_yosys_params
 INFO (00_and2_MIN_ROUTE_CHAN_WIDTH) -     tree = ET.parse(args.arch_file)
 INFO (00_and2_MIN_ROUTE_CHAN_WIDTH) -   File "/usr/lib/python3.8/xml/etree/ElementTree.py", line 1202, in parse
 INFO (00_and2_MIN_ROUTE_CHAN_WIDTH) -     tree.parse(source, parser)
 INFO (00_and2_MIN_ROUTE_CHAN_WIDTH) -   File "/usr/lib/python3.8/xml/etree/ElementTree.py", line 595, in parse
 INFO (00_and2_MIN_ROUTE_CHAN_WIDTH) -     self._root = parser._parse_whole(source)
 INFO (00_and2_MIN_ROUTE_CHAN_WIDTH) - xml.etree.ElementTree.ParseError: mismatched tag: line 298, column 3
ERROR (00_and2_MIN_ROUTE_CHAN_WIDTH) - Failed to execute openfpga flow - 00_and2_MIN_ROUTE_CHAN_WIDTH
Traceback (most recent call last):
  File "openfpga_flow/scripts/run_fpga_task.py", line 479, in run_single_script
subprocess.CalledProcessError: Command 'python3 /home/sun/OpenFPGA/openfpga_flow/scripts/run_fpga_flow.py /home/sun/OpenFPGA/ours/our_VTR_arch.xml /home/sun/OpenFPGA/pg_sample.v --top_module and2 --run_dir /home/sun/OpenFPGA/openfpga_flow/tasks/fpga_bitstream/generate_bitstream/our/run008/our_VTR_arch/and2/MIN_ROUTE_CHAN_WIDTH --fpga_flow yosys_vpr --openfpga_shell_template /home/sun/OpenFPGA/openfpga_flow/openfpga_shell_scripts/bitstream_setting_example_script.openfpga --openfpga_arch_file /home/sun/OpenFPGA/ours/our_open_arch.xml --openfpga_sim_setting_file /home/sun/OpenFPGA/openfpga_flow/openfpga_simulation_settings/fixed_sim_openfpga.xml --vpr_fpga_verilog --vpr_fpga_verilog_dir /home/sun/OpenFPGA/openfpga_flow/tasks/fpga_bitstream/generate_bitstream/our/run008/our_VTR_arch/and2/MIN_ROUTE_CHAN_WIDTH --vpr_fpga_x2p_rename_illegal_port --debug --TOP and2' returned non-zero exit status 0.

In the detailed log file: 
/home/sun/OpenFPGA/openfpga_flow/tasks/fpga_bitstream/generate_bitstream/our/run008/our_VTR_arch/and2/MIN_ROUTE_CHAN_WIDTH
it lists what openfpga has done, but I am still confused which step is wrong, and how to correct the problems. 

Thanks for your help

tangxifan, 2022-01-03:
@Sun1927 Please check your VPR architecture file. We check the XML syntax at the beginning of the flow.
The error message:
```
INFO (00_and2_MIN_ROUTE_CHAN_WIDTH) - xml.etree.ElementTree.ParseError: mismatched tag: line 298, column 3
```
indicates that there is a syntax error in LINE 298 of your VPR architecture file.

Sun1927, 2022-01-04:
thank you I will check this file.

closed 2022-01-04 by Sun1927