lnis-uofu/OpenFPGA #471: How to modify VTR architecture file

Sun1927, 2022-01-03:
Hi, 
I am trying to add additional syntax in VTR architecture file, in the openfpga document instruction, it reads "Each <pb_type> should contain a <mode> that describe the physical implementation of the <pb_type>". In our architecture, apart from io block, the others only have one operation mode, which can be omitted according to VTR architecture document. I wonder in this case if additional syntax still need to be added in each py_type when using openfpga.

Thanks for your reply

tangxifan, 2022-01-03:
@Sun1927 In this case, you do not need to explicitly specify physical mode in OpenFPGA architecture.
However, you still have specify the circuit model for each primitive ``pb_type``.

https://github.com/lnis-uofu/OpenFPGA/blob/8b06c8790c75971e881eef6c3bf46ed70c93a704/openfpga_flow/openfpga_arch/k4_N4_40nm_GlobalTile4Clk_cc_openfpga.xml#L190-L195

tangxifan, 2022-01-03:
@Sun1927 Also for questions, please try to use the discussion forum rather than issues. Issues are used to report bugs and raise feature requests. Thanks for using OpenFPGA.

Sun1927, 2022-01-04:
Thanks for your answer, that helps.

closed 2022-01-04 by Sun1927