lnis-uofu/OpenFPGA #473: some question about synthesis

kangliyu1, 2022-01-04:
Hello! I encountered a very strange problem. I synthesized a piece of code in yosys in openfpga (code is as follows code1). An error was reported during the synthesis process, as shown in Figure 1, but no error was pointed out in yosys_output.log Locally, alone, yosys found out that kill automatically exited in the corresponding place, so I am a little confused. I hope some seniors can give me some advice. Thank you very much! !

fig1:
[image]
fig2:
[image]

code1:
```
module Conv_2#(
        parameter padding       =      0                                ,
        parameter weight_bit    =      10                               ,//有符号位宽
        parameter conv_bit      =      10                               ,//无符号位宽
        parameter conv_wid      =      13                               ,
        parameter conv_heig     =      13                               ,
        parameter stride        =      1                                ,
        parameter result_len    =      11*11                            ,
        parameter in_layer      =      6                                ,
        parameter out_layer     =      16                                
)(
        input			            			                                        clk	            ,
        input			            			                                        rst	            ,
        input		    [in_layer*conv_bit-1:0]			                                conv_din	    ,
        input                                                                           din_vld         ,
        input                                                                           din_last        ,
        input           [0:1*in_layer*9*weight_bit-1]                                   kernel          ,
        output          [in_layer*out_layer*(4+weight_bit+conv_bit)/1-1:0]              conv_dout       ,
        output                                                                          busy            ,
        output          [clogb2(out_layer)-1:0]                                         kernel_addr     ,
        output  reg                                                                     kernel_vld      ,
        output  reg                                                                     dout_vld        ,
        output  reg                                                                     dout_last       

    );
        localparam rd_init1              =   3'b000       ,
                   rd_init2              =   3'b001       ,
                   rd_init3              =   3'b010       ,
                   rd_conv               =   3'b011       ; 
        reg		signed  [4+weight_bit+conv_bit-1:0]   add_res   [0:in_layer/1-1][0:out_layer-1]                 ;
        reg  	signed  [weight_bit+conv_bit-1:0]     mul_res   [0:in_layer/1-1][0:out_layer-1][0:9-1]          ;  
        wire    signed  [weight_bit-1:0]              kernel_val[0:in_layer/1*1*9-1]                            ;
        reg		signed  [conv_bit:0]                  conv_buf  [0:1-1][0:in_layer/1-1][0:9-1]                  ;        
        wire	        [in_layer*conv_bit-1:0]	      row_out	[0:3-1]                                           ;       		
        wire	        [in_layer*conv_bit-1:0]	      row_in	[0:3-1]                                           ;      
        reg		[9:0]                         cnt_row0         		            ; 		
        reg		[9:0]                         cnt_row1         		            ; 		
        reg		[9:0]                         cnt_row2         		            ; 			
        wire	[2:0]			              row_wr		                    ;
        wire	[2:0]			              row_rd		                    ;
        reg		[1:0]                         cnt_stride_w		                ;
        reg		[1:0]                         cnt_stride_h		                ;
        reg		                		      row0_rd                           ;
        reg		                		      row1_rd                           ;
        reg		                		      row2_rd                           ;
        reg                                   conv_idle                         ;
        reg                                   conv_vld                          ;
        reg                                   last_vld                          ;
        reg		[clogb2(conv_wid)-1:0]        cnt_conv              	        ;
        reg		[clogb2(out_layer)-1:0]       cnt_kernel              	        ;
        reg		[2-1:0]                       cnt_paral              	        ;
        wire	[2:0]			              row_full		                    ;
        wire	[2:0]			              row_empty		                    ;
        reg		[19:0]			              cnt_din		                    ;
        reg		[2:0]                         state_rd		                    ;
        reg		[0:0]                         row1_add_flag		                ;
        reg		[0:0]                         row0_add_flag		                ;
        reg		[clogb2(result_len)-1:0]      cnt_out                	        ;
        genvar                  a, b, c, i, j, k, l, m, n, o                    ;
        assign conv_rst		    =	din_last && din_vld && !end_din	|| rst	    ;
        assign row_wr[2]		=	din_vld	&& !row_full[2]		                ;
[... 393 lines cut ...]
        wp <= 1'b0;
    else if(!full & we)
        wp <= wp+1'b1;
    else
        wp <= wp;
end
// 读操作
always@(posedge clk or negedge rst_n)
begin
    if(!rst_n)
        dout <= {DW{1'bz}};
    else if(!empty & re)
        dout <= ram[rp];
    else
        dout <= dout;
end
//写操作
always@(posedge clk)
begin
    if(!full & we)
        ram[wp] <= din;
    else
        ram[wp] <= ram[wp];
end
endmodule

tangxifan, 2022-01-04:
@kangliyu1 
- Can you try to remove the ``-nolatches`` option?
- Can you tell which Yosys command causes the problem? According to the log file, my guess is the ``synth -top Conv_2 -flatten ``. 
- To identify if this is a bug from yosys or our tech lib, I suggest you to try other synthesis scripts through other test cases, e.g., 

https://github.com/lnis-uofu/OpenFPGA/blob/1d4fa96d547d39f0aa27b4be0de1635e2277c80d/openfpga_flow/tasks/benchmark_sweep/iwls2005/config/task.conf#L70

- If other scripts/tech lib can pass, it is a problem of the yosys script.
- If other scripts have the same problem, it could be a problem of yosys. The bug should be reported to Yosys HQ: https://github.com/YosysHQ/yosys

kangliyu1, 2022-01-05:
> @kangliyu1
>
> * Can you try to remove the `-nolatches` option?
> * Can you tell which Yosys command causes the problem? According to the log file, my guess is the `synth -top Conv_2 -flatten `.
> * To [...]

thank you for your reply!
1. It will still kill after deleting -nolatches
2. It is true that the command synth -top Conv_2 -flatten has a problem, but I have seen it for a long time and still don't know why the problem occurred.
3. Using the ys_tmpl_yosys_vpr_bram_dsp_dff_flow.ys script will show the problem of "Run yosys run failed with returncode -9" in the TECHMAP section, that is, kill here, which feels like a yosys problem, but after other codes are passed in the integrated part This error "Run yosys run failed with returncode -9" will also appear. For example, a "-9" direct interrupt (after yosys synthesis) suddenly appears in some strange places, as shown in the following pictures, but this error Is there any detailed debugging information, so I don’t know how to debug. I am really confused about this question. I hope you can reply when you have time. Thank you very much! ! By the way, does your openfpga architecture library have an architecture that supports the blif file synthesized by vivado? Do you know which circuits the yosys has restrictions on synthesis?
fig1:
[image]
fig2:
[image]
fig3:
[image]
and so on......

tangxifan, 2022-01-05:
@kangliyu1 
- The log file is often messed up due to broken pip when errors are shown. This is known. Currently I do not have a solution.
- If vivado can output blif file, we can spend some time in developing an architecture that supports the tech lib. Currently we do not have anything in stock.

tangxifan, 2022-01-05:
@kangliyu1 Can you provide the openfpga version ? (You can see it by running ``OpenFPGA/openfpga/openfpga -v``

kangliyu1, 2022-01-06:
> @kangliyu1 Can you provide the openfpga version ? (You can see it by running `OpenFPGA/openfpga/openfpga -v`
This is my openfpga version
[image]

tangxifan, 2022-01-06:
@kangliyu1 We have improved many features in OpenFPGA since 2021-09-22, including Yosys upgrade. Can you give a try on the latest master?

kangliyu1, 2022-01-06:
> We have improved many features in OpenFPGA since 2021-09-22, including Yosys upgrade. Can you give a try on the latest master?

Okay, thank you for your reply. I will try to use the upgraded version today. Does the upgraded openfpga have any more functions? There is also that vivado can generate edif files, which can run the corresponding vpr in openfpga?

tangxifan, 2022-01-06:
@kangliyu1 The major update is to bump Yosys version from 0.9 to latest 0.10 release
Also we introduce new configuration protocols.

PCF and EDIF support is scheduled this year. It will be online.

closed 2022-01-11 by kangliyu1