lnis-uofu/OpenFPGA #474: Run yosys run failed with returncode 1

Sun1927, 2022-01-04:
Hi, 
when I am trying to generate the bitstream, the results show "ERROR - -->>ERROR: Module `and2' not found!", I don't quite understand where this and module comes from, if this indicates I use the wrong configuration file? Thanks for all the help.

This is the whole result:
NFO (     MainThread) - Setting loggger in debug mode
 INFO (     MainThread) - Set up to run 2 Parallel threads
 INFO (     MainThread) - Currently running task fpga_bitstream/generate_bitstream/our
 INFO (     MainThread) - Created "run011" directory for current task run
 INFO (     MainThread) - Running "yosys_vpr" flow
 INFO (     MainThread) - Found 1 Architectures 1 Benchmarks & 1 Script Parameters
 INFO (     MainThread) - Created total 1 jobs
DEBUG (00_and2_MIN_ROUTE_CHAN_WIDTH) - Running OpenFPGA flow with [['python3', '/home/sun/OpenFPGA/openfpga_flow/scripts/run_fpga_flow.py', '/home/sun/OpenFPGA/ours/our_VTR_arch.xml', '/home/sun/OpenFPGA/ours/pg_sample.v', '--top_module', 'and2', '--run_dir', '/home/sun/OpenFPGA/openfpga_flow/tasks/fpga_bitstream/generate_bitstream/our/run011/our_VTR_arch/and2/MIN_ROUTE_CHAN_WIDTH', '--fpga_flow', 'yosys_vpr', '--openfpga_shell_template', '/home/sun/OpenFPGA/openfpga_flow/openfpga_shell_scripts/bitstream_setting_example_script.openfpga', '--openfpga_arch_file', '/home/sun/OpenFPGA/ours/our_open_arch.xml', '--openfpga_sim_setting_file', '/home/sun/OpenFPGA/openfpga_flow/openfpga_simulation_settings/fixed_sim_openfpga.xml', '--vpr_fpga_verilog', '--vpr_fpga_verilog_dir', '/home/sun/OpenFPGA/openfpga_flow/tasks/fpga_bitstream/generate_bitstream/our/run011/our_VTR_arch/and2/MIN_ROUTE_CHAN_WIDTH', '--vpr_fpga_x2p_rename_illegal_port', '--debug', '--TOP', 'and2']]
 INFO (00_and2_MIN_ROUTE_CHAN_WIDTH) - Validating command line arguments
 INFO (00_and2_MIN_ROUTE_CHAN_WIDTH) - Setting loggger in debug mode
 INFO (00_and2_MIN_ROUTE_CHAN_WIDTH) - Run directory : /home/sun/OpenFPGA/openfpga_flow/tasks/fpga_bitstream/generate_bitstream/our/run011/our_VTR_arch/and2/MIN_ROUTE_CHAN_WIDTH
 INFO (00_and2_MIN_ROUTE_CHAN_WIDTH) - Running "yosys_vpr" Flow
 INFO (00_and2_MIN_ROUTE_CHAN_WIDTH) - Extracted lut_size size from arch XML = 4
 INFO (00_and2_MIN_ROUTE_CHAN_WIDTH) - Running Yosys with lut_size = 4
 INFO (00_and2_MIN_ROUTE_CHAN_WIDTH) - Launching Run yosys 
ERROR (00_and2_MIN_ROUTE_CHAN_WIDTH) - Run yosys run failed with returncode 1
ERROR (00_and2_MIN_ROUTE_CHAN_WIDTH) - command /home/sun/OpenFPGA/yosys/yosys yosys.ys
ERROR (00_and2_MIN_ROUTE_CHAN_WIDTH) - -->>ERROR: Module `and2' not found!
ERROR (00_and2_MIN_ROUTE_CHAN_WIDTH) - Current working directory : /home/sun/OpenFPGA/openfpga_flow/tasks/fpga_bitstream/generate_bitstream/our/run011/our_VTR_arch/and2/MIN_ROUTE_CHAN_WIDTH
ERROR (00_and2_MIN_ROUTE_CHAN_WIDTH) - Failed to run Run yosys task
ERROR (00_and2_MIN_ROUTE_CHAN_WIDTH) - Exiting . . . . . .
ERROR (00_and2_MIN_ROUTE_CHAN_WIDTH) - Failed to execute openfpga flow - 00_and2_MIN_ROUTE_CHAN_WIDTH
Traceback (most recent call last):
  File "openfpga_flow/scripts/run_fpga_task.py", line 479, in run_single_script
subprocess.CalledProcessError: Command 'python3 /home/sun/OpenFPGA/openfpga_flow/scripts/run_fpga_flow.py /home/sun/OpenFPGA/ours/our_VTR_arch.xml /home/sun/OpenFPGA/ours/pg_sample.v --top_module and2 --run_dir /home/sun/OpenFPGA/openfpga_flow/tasks/fpga_bitstream/generate_bitstream/our/run011/our_VTR_arch/and2/MIN_ROUTE_CHAN_WIDTH --fpga_flow yosys_vpr --openfpga_shell_template /home/sun/OpenFPGA/openfpga_flow/openfpga_shell_scripts/bitstream_setting_example_script.openfpga --openfpga_arch_file /home/sun/OpenFPGA/ours/our_open_arch.xml --openfpga_sim_setting_file /home/sun/OpenFPGA/openfpga_flow/openfpga_simulation_settings/fixed_sim_openfpga.xml --vpr_fpga_verilog --vpr_fpga_verilog_dir /home/sun/OpenFPGA/openfpga_flow/tasks/fpga_bitstream/generate_bitstream/our/run011/our_VTR_arch/and2/MIN_ROUTE_CHAN_WIDTH --vpr_fpga_x2p_rename_illegal_port --debug --TOP and2' returned non-zero exit status 0.

tangxifan, 2022-01-04:
@Sun1927 The error indicates that the top-level module ``and2`` cannot be found in your HDL files.

In the task configuration file, you need to ensure that the top module ``bench<i>_top`` is consistent with your HDL files (``bench<i>``)
For example, 

https://github.com/lnis-uofu/OpenFPGA/blob/1d4fa96d547d39f0aa27b4be0de1635e2277c80d/openfpga_flow/tasks/fpga_bitstream/generate_bitstream/device_auto/config/task.conf#L26-L30

Sun1927, 2022-01-05:
Thanks for replying, I will modify my config file.

Sun1927, 2022-01-05:
Sorry to interrupt again, after changing the bench mark name, the openfpga flow still cannot be run, the error is like this: 
INFO - Validating command line arguments
INFO - Setting loggger in debug mode
INFO - Run directory : /home/sun/OpenFPGA/openfpga_flow/tasks/fpga_bitstream/generate_bitstream/our/run013/our_VTR_arch/and2/MIN_ROUTE_CHAN_WIDTH
INFO - Running "yosys_vpr" Flow
INFO - Extracted lut_size size from arch XML = 4
INFO - Running Yosys with lut_size = 4
INFO - Launching Run yosys 
INFO - Run yosys is written in file yosys_output.log
INFO - Launching Yosys 
INFO - Yosys is written in file yosys_rewrite.log
INFO - Running OpenFPGA Shell Engine 
INFO - Launching OpenFPGA Shell Run 
INFO - OpenFPGAShell Revision: a42342fa Compiled: 2021-11-03T19:10:13
ERROR - OpenFPGA Shell Run run failed with returncode 1
ERROR - command /home/sun/OpenFPGA/openfpga/openfpga -batch -f and2_run.openfpga
ERROR - Current working directory : /home/sun/OpenFPGA/openfpga_flow/tasks/fpga_bitstream/generate_bitstream/our/run013/our_VTR_arch/and2/MIN_ROUTE_CHAN_WIDTH
ERROR - Failed to run OpenFPGA Shell Run task
ERROR - Exiting . . . . . .

Is this because we need to set min route chan width?

tangxifan, 2022-01-05:
@Sun1927 For detailed errors, please check the openfpgashell.log under your runtime directory:

```
/home/sun/OpenFPGA/openfpga_flow/tasks/fpga_bitstream/generate_bitstream/our/run013/our_VTR_arch/and2/MIN_ROUTE_CHAN_WIDTH
```

tangxifan, 2022-01-18:
@Sun1927 Let me know if you have any questions. I close it for now.

closed 2022-01-18 by tangxifan