lnis-uofu/OpenFPGA #477: some question about synth

kangliyu1, 2022-01-06:
Hello! After I successfully integrated using yosys alone, I put the relevant code in openfpga and ran the entire process, and found that the error in the picture below appeared in yosys_rewrite.log. The relevant code was integrated in vivado and passed. I don’t understand why this occurs. This kind of problem, after checking the code, I didn’t find the related problem of "out_of_range", so I would like to ask the seniors, I hope you can give some pointers, thank you very much! !
fig1:
[image]

tangxifan, 2022-01-06:
@kangliyu1 Rewrite is mainly for verification purpose. You can bypass it if you just want to run bitstream generation flows.

kangliyu1, 2022-01-07:
> @kangliyu1 Rewrite is mainly for verification purpose. You can bypass it if you just want to run bitstream generation flows.

Thank you for your reply, can I only bypass this part by commenting the rewrite part of the run_fpga_flow.py and run_fpga_task.py code? What function does this part of verification mainly verify? If an error occurs, will the subsequent content be wrong?