lnis-uofu/OpenFPGA #480: some question about synth

kangliyu1, 2022-01-11:
Hello! I have a question I want to ask you, that is, when synthesizing a simple verilog (code1), I found a problem as follows, I hope the teacher can give some pointers, thank you very much! !
Case 1: Use the script OpenFPGA/openfpga_flow/misc/ys_tmpl_yosys_vpr_bram_dsp_dff_flow.ys to synthesize it, the result of the synthesis is shown in Figure 1, but an error is reported during the vpr process, as shown in Figure 2, the displayed error is that there is no PMUX module , but no matter how I modify this script pmux will be synthesized, I don't know where to modify it, so I want to ask you how to solve it

code1:
```
module RAM(qp,qtp,qw,dp,dtp,dw,write_addressp,read_addressp,write_addresstp,read_addresstp,write_addressw,read_addressw,we_p,we_tp,we_w,re_p,re_tp,re_w,clk);
parameter picture_size=28;
parameter SIZE_1=11;
parameter SIZE_2=22;
parameter SIZE_4=44;
parameter SIZE_9=99;
parameter SIZE_address_pix=13;
parameter SIZE_address_pix_t=12;
parameter SIZE_address_wei=9;

output reg signed [SIZE_1-1:0] qp;       //read data
output reg signed [(SIZE_2)*1-1:0] qtp;       //read data
output reg signed [SIZE_9-1:0] qw;      //read weight
input signed [SIZE_1-1:0] dp;   //write data
input signed [(SIZE_2)*1-1:0] dtp;   //write data
input signed [SIZE_9-1:0] dw;   //write weight
input [SIZE_address_pix-1:0] write_addressp, read_addressp;
input [SIZE_address_pix_t-1:0] write_addresstp, read_addresstp;
input [SIZE_address_wei-1:0] write_addressw, read_addressw;
input we_p,we_tp,we_w,re_p,re_tp,re_w,clk;

reg signed [SIZE_1-1:0] mem [0:picture_size*picture_size*8+picture_size*picture_size-1];
reg signed [(SIZE_2)*1-1:0] mem_t [0:picture_size*picture_size*4-1];
reg signed [SIZE_9-1:0] weight [0:256];
always @ (posedge clk)
    begin
        if (we_p) mem[write_addressp] <= dp;
		if (we_tp) mem_t[write_addresstp] <= dtp;
		if (we_w) weight[write_addressw] <= dw;
    end
always @ (posedge clk)
    begin
        if (re_p) qp <= mem[read_addressp];
		if (re_tp) qtp <= mem_t[read_addresstp];
        if (re_w) qw <= weight[read_addressw];
    end
endmodule
```
fig1:
[image]

fig2:
[image]

Case 2: Use the default script to synthesize code1 (the default script is shown in script 1)
The result is an extremely large-scale synthesis of a simple circuit, as shown in Figure 3:
script 1:
```
# Yosys synthesis script for RAM
# Read verilog files
read_verilog -nolatches ./benchmark/CNN_FPGA.v
# Technology mapping
hierarchy -top RAM
proc
techmap -D NO_LUT -map +/adff2dff.v
# Synthesis
synth -top RAM -flatten
clean
# LUT mapping
abc -lut 6
# Check
synth -run check
# Clean and output blif
opt_clean -purge
write_blif RAM_yosys_out.blif
```
fig3:
[image]

kangliyu1, 2022-01-12:
> @kangliyu1
> 
> * If you use the default script, it is designed to map to LUT and FFs.  So if you expect BRAM and DSP, it will produce a large resource usage.
> * The problem on the $pmux is mainly due to Yosys failed to techmap some cells. So the internal cells are revealed. My suggestion is to read the yosys user manual (understand how pmux works), and then tweak the yosys script.

ok, i'm learning, thanks for your reply

closed 2022-01-12 by kangliyu1

code-joswin, 2025-06-27:
Hi,
I faced a similar problem. I don't know if it's a little too late to comment  but adding a generic techmap comment after all the mappings but before the abc mapping helped me resolve this issue. As far as I understand the memory map step introduces this $pmux which stays as is unless you ask the tool to resolve it by adding techmap command again.
 
Here is how my script looks at the end:
```
#########################
# Map flip-flops
#########################
dfflegalize -cell $_DFF_P_ 0
techmap -map +/adff2dff.v
log "Statistics after dff map"
stat -width
#techmap -D NO_ENABLE -map +/adff2dff.v  # Strip enables/resets
opt_expr -mux_undef
simplemap
opt_expr
opt_merge
opt_dff -nodffe -nosdff
opt_clean
opt -nodffe -nosdff

techmap  #Add a generic techmap pass after architecture-specific mapping to decompose all hierarchies. To avoid pmux for spmv rtl

# After reading and synthesizing your design
#echo "===== PRE-MAPPING GATE STATISTICS ====="
log "Statistics before LUT mapping"
stat -width

#########################
# Map LUTs
#########################
abc -lut ${LUT_SIZE}
```

Thanks.