lnis-uofu/OpenFPGA #532: Some questions about dff mapping

kangliyu1, 2022-02-15:
Hello! I encountered several problems in the process of using openfpga,
1. When using script 1 for synthesis, the result is as shown in Figure 1, that is, _DFF_NP0_ and _DFF_NP1_ appear, but in k6_frac_N1o_tileabl...racff_40nm_dff_map.v
These two DFF modules are not found in this file, so the VTR part cannot pass, so I would like to ask you, can I build the two modules _DFF_NP0_ and _DFF_NP1_ in k6_frac_N1o_tileabl...racff_40nm_dff_map.v by myself? (As shown in Figure 2), the VTR can indeed pass after this construction, but I am worried that it will affect other parts, so I still want to ask if there are still places that need to be modified or if this modification is not acceptable.
2. The __soft_mul module also appears in the synthesis result, and this module cannot perform the corresponding VTR part. How should this be handled? Is it possible to use chtype -map $__soft_mul $mul to change his type? Is there a better way?
fig1:
[image]

script:
```
# Yosys synthesis script for ${TOP_MODULE}
#########################
# Parse input files
#########################
# Read verilog files
${READ_VERILOG_FILE}
# Read technology library
read_verilog -lib -specify ${YOSYS_CELL_SIM_VERILOG}
#########################
# Prepare for synthesis
#########################
# Identify top module from hierarchy
hierarchy -check -top ${TOP_MODULE}
# - Convert process blocks to AST
proc
# Flatten all the gates/primitives
flatten
# Identify tri-state buffers from 'z' signal in AST
# with follow-up optimizations to clean up AST
tribuf -logic
opt_expr
opt_clean
# demote inout ports to input or output port
# with follow-up optimizations to clean up AST
deminout
opt
opt_expr
opt_clean
check
opt
wreduce -keepdc
peepopt
pmuxtree
opt_clean
########################
# Map multipliers
# Inspired from synth_xilinx.cc
#########################
# Avoid merging any registers into DSP, reserve memory port registers first
memory_dff
wreduce t:$mul
techmap -map +/mul2dsp.v -map ${YOSYS_DSP_MAP_VERILOG} ${YOSYS_DSP_MAP_PARAMETERS}
select a:mul2dsp
setattr -unset mul2dsp
opt_expr -fine
wreduce
select -clear
chtype -set $mul t:$__soft_mul# Extract arithmetic functions

#########################
# Run coarse synthesis
#########################
# Run a tech map with default library
techmap
alumacc
share
opt
fsm
# Run a quick follow-up optimization to sweep out unused nets/signals
opt -fast
# Optimize any memory cells by merging share-able ports and collecting all the ports belonging to memorcy cells  
memory -nomap
opt_clean

#########################
# Map logics to BRAMs
#########################
memory_bram -rules ${YOSYS_BRAM_MAP_RULES}
techmap -map ${YOSYS_BRAM_MAP_VERILOG}
opt -fast -mux_undef -undriven -fine
memory_map
opt -undriven -fine
pmuxtree
#########################
# Map flip-flops
#########################
techmap -map ${YOSYS_DFF_MAP_VERILOG}
opt_expr -mux_undef
simplemap
opt_expr
opt_merge
opt_rmdff
opt_clean
opt

#########################
# Map LUTs
#########################
abc -lut ${LUT_SIZE}

#########################
# Check and show statisitics
#########################
hierarchy -check
stat
#########################
# Output netlists
#########################
opt_clean -purge
write_blif ${OUTPUT_BLIF}
```

fig2:
[image]

tangxifan, 2022-02-15:
@kangliyu1 Whether to add ``DFF_NP0`` and ``DFF_NP1`` depends on if your architecture can support it or not. If I remember correctly, the architecture you showed supports these. That is why they are there.

Regarding your synthesis results, I think the synthesis flow failed to map some logics. That is why you see these internal modules whose names start with ``$``. You may need to optimize your synthesis scripts.

kangliyu1, 2022-02-16:
> 2\. chtype -map $__soft_mul $mul

It seemed that it was not supported at first, so I added two structures, DFF_NP0 and DFF_NP1, in Figure 2. I don’t know if there will be other bad effects. After I added these two modules, the synthesis can be passed.

$__soft_mul is indeed not mapped, because there is no such module, and the relevant description of this module has not been found, and then after the Map multipliers are deleted, it is not mapped to this module, but I need Map multipliers, so I have been adjusting it, and finally By adding "chtype -map $__soft_mul $mul" the script can be passed, but this is the statement that changes __soft_mul to mul. It feels that it will have some impact on the follow-up, so I want to ask if you have a better way,
一开始好像是不支持的，所以我在图二那里添加了DFF_NP0 和 DFF_NP1这两个结构，不知道是否会出现其他不好的影响，我加了这两个模块之后确实是综合可以通过了。

$__soft_mul确实是没能映射，因为没有这个模块，同时也没找到这个模块的相关描述，然后把Map multipliers删除之后就没有映射到这个模块了，但是我需要Map multipliers，所以一直在调整，最后是通过添加”chtype -map $__soft_mul $mul“脚本导致的可以通过，但是这个是将__soft_mul 变为mul的语句感觉会对后续产生一些影响所以想问问您是否有更好的办法，

closed 2022-02-22 by kangliyu1