lnis-uofu/OpenFPGA #551: Some questions about the resulting FPGA-Verilog

kangliyu1, 2022-02-22:
Hello, after I used python3 openfpga_flow/scripts/run_fpga_task.py benchmark_sweep/mac_units to successfully generate a 25X30 FPGA soft core, and successfully mapped the relevant circuit into the FPGA soft core, the relevant bitstream was generated, but during the simulation There are many problems when using the generated TOP_autocheck_top_tb.v to perform the related simulation, the warning of the zero delay loop appears (as shown in Figure 1), and after modifying the testbench (as shown in code1), the zero delay still appears. The warning of the loop, is there a problem with the soft core of the generated FPGA? How can I solve this problem, I hope you can give some advice, thank you very much! !
fig1:
[image]
code1:
[image]
[image]
[image]

tangxifan, 2022-02-22:
@kangliyu1 Can you tell which commit you are using?

If you see zero-delay, try to ensure that you turn on the UNIT_DELAY in your standard cell library.
Just like the ``--include_timing`` option:

https://openfpga.readthedocs.io/en/master/manual/openfpga_shell/openfpga_commands/fpga_verilog_commands/#cmdoption-include_timing

Also, try the options when generating testbenches
- https://openfpga.readthedocs.io/en/master/manual/openfpga_shell/openfpga_commands/fpga_verilog_commands/#cmdoption-include_signal_init
- https://openfpga.readthedocs.io/en/master/manual/openfpga_shell/openfpga_commands/fpga_verilog_commands/#cmdoption-include_signal_init

kangliyu1, 2022-02-28:
> 

Ok thank you for your reply, I added it, but there is still a problem, but I adjusted the emulation tcl script of VCS to solve this problem

closed 2022-02-28 by kangliyu1