lnis-uofu/OpenFPGA #562: Building make Error on Centos7

msaideroglu, 2022-02-26:
I'm getting this error on my centos 7 machine when "make all" command was running. I upgraded gcc version to 7.3, it still give this error.

"[  1%] Building CXX object libs/EXTERNAL/libargparse/CMakeFiles/libargparse.dir/src/argparse.cpp.o
In file included from /home/synopsys/OpenFPGA/libs/EXTERNAL/libargparse/src/argparse_default_converter.hpp:8:0,
                 from /home/synopsys/OpenFPGA/libs/EXTERNAL/libargparse/src/argparse.hpp:12,
                 from /home/synopsys/OpenFPGA/libs/EXTERNAL/libargparse/src/argparse.cpp:9:
/home/synopsys/OpenFPGA/libs/EXTERNAL/libargparse/src/argparse_value.hpp: In constructor ‘constexpr argparse::ArgValue<bool>::ArgValue()’:
/home/synopsys/OpenFPGA/libs/EXTERNAL/libargparse/src/argparse_value.hpp:44:11: error: conversion from ‘const char [1]’ to non-scalar type ‘std::string {aka std::basic_string<char>}’ requested
     class ArgValue {
           ^
/home/synopsys/OpenFPGA/libs/EXTERNAL/libargparse/src/argparse_value.hpp:44:11: error: conversion from ‘const char [1]’ to non-scalar type ‘std::string {aka std::basic_string<char>}’ requested
/home/synopsys/OpenFPGA/libs/EXTERNAL/libargparse/src/argparse.cpp: In constructor ‘argparse::ArgumentParser::ArgumentParser(std::string, std::string, std::ostream&)’:
/home/synopsys/OpenFPGA/libs/EXTERNAL/libargparse/src/argparse.cpp:22:17: note: synthesized method ‘constexpr argparse::ArgValue<bool>::ArgValue()’ first required here 
         , os_(os)
                 ^
make[3]: *** [libs/EXTERNAL/libargparse/CMakeFiles/libargparse.dir/src/argparse.cpp.o] Error 1
make[3]: Leaving directory `/home/synopsys/OpenFPGA/build'
make[2]: *** [libs/EXTERNAL/libargparse/CMakeFiles/libargparse.dir/all] Error 2
make[2]: Leaving directory `/home/synopsys/OpenFPGA/build'
make[1]: *** [all] Error 2
make[1]: Leaving directory `/home/synopsys/OpenFPGA/build'
make: *** [all] Error 2
[  1%] Building CXX object libs/EXTERNAL/libargparse/CMakeFiles/libargparse.dir/src/argparse.cpp.o
In file included from /home/synopsys/OpenFPGA/libs/EXTERNAL/libargparse/src/argparse_default_converter.hpp:8:0,
                 from /home/synopsys/OpenFPGA/libs/EXTERNAL/libargparse/src/argparse.hpp:12,
                 from /home/synopsys/OpenFPGA/libs/EXTERNAL/libargparse/src/argparse.cpp:9:
/home/synopsys/OpenFPGA/libs/EXTERNAL/libargparse/src/argparse_value.hpp: In constructor ‘constexpr argparse::ArgValue<bool>::ArgValue()’:
/home/synopsys/OpenFPGA/libs/EXTERNAL/libargparse/src/argparse_value.hpp:44:11: error: conversion from ‘const char [1]’ to non-scalar type ‘std::string {aka std::basic_string<char>}’ requested
     class ArgValue {
           ^
/home/synopsys/OpenFPGA/libs/EXTERNAL/libargparse/src/argparse_value.hpp:44:11: error: conversion from ‘const char [1]’ to non-scalar type ‘std::string {aka std::basic_string<char>}’ requested
/home/synopsys/OpenFPGA/libs/EXTERNAL/libargparse/src/argparse.cpp: In constructor ‘argparse::ArgumentParser::ArgumentParser(std::string, std::string, std::ostream&)’:
/home/synopsys/OpenFPGA/libs/EXTERNAL/libargparse/src/argparse.cpp:22:17: note: synthesized method ‘constexpr argparse::ArgValue<bool>::ArgValue()’ first required here 
         , os_(os)
                 ^
make[3]: *** [libs/EXTERNAL/libargparse/CMakeFiles/libargparse.dir/src/argparse.cpp.o] Error 1
make[3]: Leaving directory `/home/synopsys/OpenFPGA/build'
make[2]: *** [libs/EXTERNAL/libargparse/CMakeFiles/libargparse.dir/all] Error 2
make[2]: Leaving directory `/home/synopsys/OpenFPGA/build'
make[1]: *** [all] Error 2
make[1]: Leaving directory `/home/synopsys/OpenFPGA/build'
make: *** [all] Error 2"

tangxifan, 2022-03-01:
Base on what I see, if CentOS8 is used, we cannot guarantee compilation compatibility now.
Currently, OpenFPGA's compilation compatibility is continuously checked on the following settings

 
[image]

I think it is a good idea to put CentOS compilation compatibility in CI.

https://github.com/OPM/ResInsight/blob/dev/.github/workflows/centos7.yml

mithro, 2022-03-01:
BTW Centos 8 was End of Lifed in December 2021 (see https://www.centos.org/centos-linux-eol/) but surprisingly Centos 7 will get maintenance updates until 2024-06-30!

msaideroglu, 2022-03-01:
Bur my system is Centos 7. Which gcc version should I used? Is the problem that is above because of git or gcc version?

msaideroglu, 2022-03-05:
I almost use same system properties with the compilation guideline video on youtube:
[image]
However it gives errors like this:
[image]
How can I solve this issue?

closed 2022-03-23 by msaideroglu