lnis-uofu/OpenFPGA #565: some question about FPGA-Verilog

kangliyu1, 2022-02-28:
Hello, using the openfpga platform to generate an EFPGA module with a 24*24 array, the layout is: one column of CLB\one column of BRAM, one column\DSP\one column of CLB, this is repeated six times for a total of 24 columns, each column has 24 unit spaces, in the process of simulation I have some questions and I want to ask you:
1. When using VCS to perform the benchmark simulation of mac_2, it is found that the PAD configured as the output of the mapping circuit has always maintained a high-impedance state (as shown in Figure 1). In the process of looking for signal problems, it was found that the generated FPGA-Verilog may have some Problem (for example: the output port of my configuration is gfpga_pad_GPIO_PAD[518], the incoming signal finally found that it was driven by the PAD port of the GPIO_0_ submodule of the logical_tile_io_mode_physical_iopad_0 submodule of the logical_tile_io_mode_io__6 submodule of the grid_io_bottom_8__0_ module, so it continued to track the PAD signal It is found that the PAD is jointly determined by the DIR and A signals. At this time, it is found that the A signal is always in a high-impedance state. Tracking forward, it is found that A is determined by A<-iopad_outpad<-direct_interc_1_out<-io_outpad<-top_width_0_height_0_subtile_6__pin_outpad_0_ <-grid_io_bottom_8__0__undriven_top_width_0_height_0_subtile_6__pin_outpad_0_, but only grid_io_bottom_8__0__undriven_top_width_0_height_0_subtile_6__pin_outpad_0_ "wire [0: 0] grid_io_bottom_8__0__undriven_top_width_0_height_0_subtile_6__pin_outpad_0_;" FPGA_TOP module in this definition, and no other logical, nor is the output of other modules, so the sum found wire [0: 0] grid_io_bottom_8__0__undriven_top_width_0_height_0_subtile_6__pin_outpad_0_; The signal input to the grid_io_bottom_8__0_ module is just a wire signal, and no other logic has been found to cause it to remain in a high-impedance state, resulting in the mapping circuit failing to successfully implement related functions.
To sum up, I would like to ask which step you are generating fpga-verilog caused by? How can I go about solving this problem, I hope you can give some advice, thanks a lot! !
2. Last time, you suggested that you can use fabric key to customize your configuration chain. After I printed out the distribution of bitstream, how can I use this data to customize the most suitable configuration chain? Does it mean that the number of code streams that need to be configured for each block will be faster on average? I hope you can give some pointers, thank you very much! ! ! !

[image]

您好，在使用openfpga平台生成24*24阵列的EFPGA模块，布局是：一列CLB\一列BRAM一列\DSP\一列CLB这样重复六次共24列，每列24个单元空间，在仿真的过程中遇到了一些问题想要向您请教一下：
1、在使用VCS对其进行mac_2的benchmark仿真时发现配置为映射电路输出的PAD一直保持高阻状态（如图1所示），在找信号问题的过程中发现可能是生成的FPGA-Verilog有些问题（例如：我的配置的输出端口是gfpga_pad_GPIO_PAD[518]，追踪信号最终发现他是由grid_io_bottom_8__0_模块的logical_tile_io_mode_io__6子模块的logical_tile_io_mode_physical__iopad_0子模块的GPIO_0_子模块的PAD端口驱动的，于是继续追踪PAD信号的产生逻辑，发现PAD是由DIR和A信号共同决定的，在此时发现A信号一直是保持为高阻状态，往前追踪发现A是由A<-iopad_outpad<-direct_interc_1_out<-io_outpad<-top_width_0_height_0_subtile_6__pin_outpad_0_<-grid_io_bottom_8__0__undriven_top_width_0_height_0_subtile_6__pin_outpad_0_，而grid_io_bottom_8__0__undriven_top_width_0_height_0_subtile_6__pin_outpad_0_在FPGA_TOP模块中只有“wire [0:0] grid_io_bottom_8__0__undriven_top_width_0_height_0_subtile_6__pin_outpad_0_;”这个定义，并没有其他的逻辑，也不是其他模块的输出，所以综上发现wire [0:0] grid_io_bottom_8__0__undriven_top_width_0_height_0_subtile_6__pin_outpad_0_;这个输入到grid_io_bottom_8__0_模块的信号只是一个wire信号，且没有发现其他的逻辑导致了其一直保持高阻状态，导致了映射电路没有成功实现相关的功能。
综上想请问下您这是在生成fpga-verilog哪一步导致的呢？我该如何去解决这个问题，希望您能给些建议万分感谢！！

2、您上次建议可以用 fabric key来定制化自己的configuration chain，在我打印出了bitstream的分布情况之后，我该如何去利用这个数据去定制最适合的configuration chain呢？是说每一块需要配置的码流数量越平均就会越快吗？希望您能指点一下，万分感谢！！！！

tangxifan, 2022-03-01:
- You may need to check the HDL of your configuration memory, such as DFF. In some HDLs of OpenFPGA, it is designed for preconfigured testbenches: 
https://github.com/lnis-uofu/OpenFPGA/blob/27fd7b0fe791cd2e6f1c13e1cd32f5bf4042d1ce/openfpga_flow/openfpga_cell_library/verilog/latch.v#L28-L34

It may cause some issues when you run full testbenches.

- You are right. Configuration time of configuration chain is bounded by the longest chain

- If you have undriven signals, please check your openfpga shell script. See if you have turn off the ``--frame_view`` when calling the ``build_fabric`` command

Last but not least, when reporting an issue, please use our issue template. We need version number, test cases and related information, without which it is hard to diagnose.

kangliyu1, 2022-03-03:
> * You may need to check the HDL of your configuration memory, such as DFF. In some HDLs of OpenFPGA, it is designed for preconfigured testbenches:

Thank you for your reply, sorry for not using your template to ask questions, next time I will use your template to ask questions;
1. I turned off --frame_view when I called the build_fabric command, but there is still an undriven signal. Will it have any effect? (As shown in Figure 1 and 2)
2. When I use modelsim to simulate, I find that the error as shown in Figure 3 will appear. After changing the settings of modelsim, it will still appear. Does this need to be changed?
fig1:
[image]

fig2:
[image]

fig3:
[image]

tangxifan, 2022-06-01:
Undriven nets on cin and cout are normal in current architectures. They are not connected to any block when CLBs are placed in the 1st and last in a column

closed 2022-06-01 by tangxifan