lnis-uofu/OpenFPGA #574: some question about simulation

kangliyu1, 2022-03-03:
Hello, I used the testbench generated by the openfpga platform to simulate after mapping the encryption module of openfpga_flow/benchmarks/iwls2005/aes_core to the FPGA module I generated, and found that the simulation results were wrong (as shown in Figure 1, Figure 2 is vivado The simulation result), many signals will appear unknown state (X state), how did you build the testbench of this routine, but the whole process of openfpga did not report an error, so I would like to ask you when you are testing these routines? How to deal with these problems, how to locate the problem of the FPGA module or the problem of test excitation? How to locate the problem?
fig1:
[image]

fig2:
[image]

closed 2022-03-24 by kangliyu1

zliu1Charlotte, 2022-09-20:
I port the netlist into vivado and my gfpga outputs are all 'z', do you remember how to resolve this issue?

kangliyu1, 2022-09-21:
sorry, i didn't fix it

Send time: Wednesday, Sep 21, 2022 1:54 AM
Cc: "さ挚さ挚 ***@***.***&gt;; "State ***@***.***&gt;;