lnis-uofu/OpenFPGA #597: Simulation fails when FPGA architecture is changed

yunuseryilmaz18, 2022-03-21:
> **Describe the bug**
> I've generated fabric, testbench, and bitstream with k4_frac_N4_tileable_40nm architecture and 8-bit counter with sync reset as benchmark. The simulation concludes successfully. However, the simulation fails when I used the same benchmark files and k6_frac_N10_40nm architecture. 

> **Which part of OpenFPGA is buggy**
> [X] FPGA-Verilog 

> **To Reproduce**
> Steps to reproduce the behaviour:
> 1. Clone OpenFPGA repository and checkout commit id: 9f65ebdef9cc72c8c33e72c3d6f8171121d9f881
> 2. Execute generate_fabric and generate_testbench tasks with k4_frac_N4_tileable_40nm architecture and BLIF and ACT files for 8-bit counter with sync reset.
> 3. Run the testbench on modelsim and wait until it's completed.
> 4. Execute generate_fabric and generate_testbench tasks with k6_frac_N10_tileable_40nm architecture and the same BLIF and ACT files for 8-bit counter with sync reset in Step 2.
> 5. Run the testbench on modelsim and wait until it's completed.
> 6. See the error.

> **Expected behaviour**
> I expected to see a successful simulation of the counter circuit with k6_frac_N10_tileable_40nm architecture just as the simulation that had run on k4_frac_N4_tileable_40 nm architecture.

> **Screenshots**
> [image]
> [image]
> The waveforms of the simulation and the console output with k4_frac_N4_tileable_40nm.
> [image]
> [image]
> The waveforms of the simulation and the console output with k6_frac_N10_tileable_40nm.

> **Enviornment (please complete the following information):**

>  - OS: Ubunu 20.04

>  - Compiler: gcc-9.4 and Python 3.8.10

> - Version: Current master

> **Additional context**
> The benchmark files that I've used are given below:
[counter.zip](https://github.com/lnis-uofu/OpenFPGA/files/8315025/counter.zip)

mattvlsi, 2022-04-18:
Hi @yunuseryilmaz18. My simulation fails with k4_frac_N4_tileable_40nm architecture, too. I also get port instantiation problem like this:
`iverilog -o compiled_counter ./SRC/counter_include_netlists.v -s counter_autocheck_top_tb -I ./SRC
././SRC/counter_autocheck_top_tb.v:158: warning: Port 6 (gfpga_pad_GPIO_PAD) of fpga_top expects 32 bits, got 64.
././SRC/counter_autocheck_top_tb.v:158:        : Leaving 32 high bits of the expression dangling.
././SRC/counter_autocheck_top_tb.v:256: error: port ``result_0_'' is not a port of REF_DUT.
././SRC/counter_autocheck_top_tb.v:256: error: port ``result_1_'' is not a port of REF_DUT.
././SRC/counter_autocheck_top_tb.v:256: error: port ``result_2_'' is not a port of REF_DUT.
././SRC/counter_autocheck_top_tb.v:256: error: port ``result_3_'' is not a port of REF_DUT.
././SRC/counter_autocheck_top_tb.v:256: error: port ``result_4_'' is not a port of REF_DUT.
././SRC/counter_autocheck_top_tb.v:256: error: port ``result_5_'' is not a port of REF_DUT.
././SRC/counter_autocheck_top_tb.v:256: error: port ``result_6_'' is not a port of REF_DUT.
././SRC/counter_autocheck_top_tb.v:256: error: port ``result_7_'' is not a port of REF_DUT.
8 error(s) during elaboration.
8`

When I change instantiation as:
`	counter REF_DUT(
		clk,
		reset_shared_input,
		result_0__benchmark,
		result_1__benchmark,
		result_2__benchmark,
		result_3__benchmark,
		result_4__benchmark,
		result_5__benchmark,
		result_6__benchmark,
		result_7__benchmark
	);`

It compiles and simulates however simulation fails in several points. Output result IO's doesn't count. Did you do a configuration in testbench to run it successfully? If possible can you share your counter_autocheck_testbench file for k4_frac_N4_tileable arch? Thanks.

tangxifan, 2022-06-01:
@yunuseryilmaz18 Can you share the openfpga task configuration file?

yunuseryilmaz18, 2022-06-13:
@tangxifan Hi, I've never tried again after I've opened the issue. Today, I tried once again and I saw the correct results. I will close the issue. Thank you.

closed 2022-06-13 by yunuseryilmaz18