lnis-uofu/OpenFPGA #626: Synthesizable Netlist Generation

mattvlsi, 2022-04-19:
I saw that we can create synthesizable netlist via Fpga-Verilog that can be run into DC or some other synthesis tools in a webinar. Is it a different netlist type than generate_fabric task generate? If yes, how can I generate it? Additionally, which one is more robust for PnR applications? Generating synthesizable netlist and run into  synthesis tools and or generating tech-mapped netlist and run into PnR tools directly?

ganeshgore, 2022-04-21:
Hello @mattvlsi 

Thank you for using OpenFPGA.
You can definitely generate an RTL design from OpenFPGA, you needs to set `dump_structural_verilog="true"` for all your circuit models. which will force OpenFPGA to write structural Verilog for that design 
https://openfpga.readthedocs.io/en/latest/manual/arch_lang/circuit_library/?highlight=dump_structural_verilog#cmdoption-arg-circuit_model

Alternately, you can also take a tech-mapped netlist and flatten it in any synthesis tool to re-synthesize based on your preferred constraint. 

Which one is better? My opinion is that an externally synthesized netlist will have a smaller area and better performance, but to achieve that, you need to create proper SDCs manually (blockwise). 
The other reason why we use tech mapped netlist as of now is to enable pre-configured testbenches (which requires the exact hierarchical path of each configuration flipflop) and timing extraction for VPR back annotation  (which requires a definite boundary for sub-modules like LUT, crossbar, etc.)

mattvlsi, 2022-04-22:
Thanks @ganeshgore

closed 2022-04-22 by mattvlsi