lnis-uofu/OpenFPGA #631: Running a basic multi-bit code

A-Sathvik, 2022-04-29:
I am able to execute the basic and, or gates using the default architectures, and default settings of the task configuration file. But, when I try to declare a 2-bit output/wire (output [1:0] and_out;), Synthesis is not successful or the flow gets stopped during simulation.

Am I missing any obvious thing while changing one of the wire/output to vector in the task.conf file or somewhere?

ganeshgore, 2022-05-03:
Hello @A-Sathvik ,
If you have a bus port in the design, please remove `--explicit_port_mapping`  from any of the following commands

- write_full_testbench
- write_preconfigured_fabric_wrapper
- write_preconfigured_testbench

A-Sathvik, 2022-05-03:
The problem is solved!!! Thank you @ganeshgore Sir

A-Sathvik, 2022-05-04:
Another query,
How do we specify the number of CLBs or tiles we require in the VPR architecture?
For example, what changes in the XML code do we have to make to get FPGA fabric with only 6x6 tiles?

ganeshgore, 2022-05-04:
You can refer to Verilog-to-routing documentation for that
https://docs.verilogtorouting.org/en/latest/arch/reference/#tag-%3Cfixed_layoutname=%22string%22width=%22int%22height=%22int%22%3E

this documentation details the complete XML modeling language for the FPGA

A-Sathvik, 2022-05-05:
For example, considering https://github.com/lnis-uofu/OpenFPGA/blob/master/openfpga_flow/vpr_arch/k4_frac_N4_tileable_40nm.xml VPR architecture

How do we know/specify how many basic tile locations can the FPGA architecture accommodate?

ganeshgore, 2022-05-05:
You can refer to the documentation I shared above that details everything about the architecture modelling.

The architecture file your shared has 3 configuration 
- automated (Scales automatically based on the benchmark size)
- 2x2 and,
-  4x4