lnis-uofu/OpenFPGA #645: some question about FPGA-VERILOG

kangliyu1, 2022-05-07:
> **Describe the bug**
> A clear and concise description of what the bug is.
Hello! Recently, I wanted to get some results after the synthesis of the FPGA module generated by Opefpga, so there were a lot of warnings in the process of using DC, which made it impossible to run and finally kill, so imagine you ask what you used for synthesis before. What does the tool do? Or what tools did you use to get those reports on area power consumption? Does Openfpga have this function?Or would you like to ask if you have modified anything before synthesizing? I hope you can give some pointers, thank you very much

> **To Reproduce**
> Steps to reproduce the behavior:
> 1. Clone OpenFPGA repository and checkout commit id:
Version: 1.1.53-dev+968a03cd-dirty

> 2. Execute OpenFPGA task or your own example: 
python3 openfpga_flow/scripts/run_fpga_task.py basic_tests/full_testbench/configuration_chain
> 3. See error
none

tangxifan, 2022-06-01:
@kangliyu1 

Since there are many comb loops in FPGA fabric, ASIC synthesis tools have difficulties when optimizing.  
Base on our previous practice, we typically use technology mapped netlists (skip synthesis).
We are now actively looking into synthesis support.
We will keep you posted on new features.

closed 2022-06-01 by tangxifan

Gurusatwik, 2024-04-11:
Do you mean that your practice is to skip synthesis in OpenFPGA Flow? If YES, then what file are you giving as input to the Semi-custom design flow?