lnis-uofu/OpenFPGA #657: Custom block mapping

cristianomodesti, 2022-05-17:
Good morning,
I am trying to build a custom FPGA architecture with some DSP blocks in it but I am not able to make VPR map my test into one of those custom blocks. 
As a first test, I am trying to map a 2-bit multiplier into a custom tile (`clb_mul16x16`) in the FPGA architecture. 
So I described both VPR and OpenFPGA xml architecture description adding the following:
##### VPR xml
```
<models>
  ...
  <model name="mul16x16">
    <input_ports>
      <port name="in1" combinational_sink_ports="out"/>
      <port name="in2" combinational_sink_ports="out"/>
    </input_ports>
    <output_ports>
      <port name="out"/>
    </output_ports>
  </model>
</models>

<tiles>
  ...
  <tile name="clb_mul16x16" area="53894">	
    <equivalent_sites>
      <site pb_type="clb_mul16x16"/>
    </equivalent_sites>
    <input name="in1" num_pins="2" equivalent="none"/>
    <input name="in2" num_pins="2" equivalent="none"/>
    <output name="out" num_pins="4" equivalent="none"/>
    <clock name="clk" num_pins="1"/>
    <fc in_type="frac" in_val="0.5" out_type="frac" out_val="0.5"/>
    <pinlocations pattern="custom">
      <loc side="left">clb_mul16x16.in1</loc>
      <loc side="top">clb_mul16x16.in2 clb_mul16x16.clk</loc>
      <loc side="right">clb_mul16x16.out[3:2]</loc>
      <loc side="bottom">clb_mul16x16.out[1:0]</loc>
    </pinlocations>
  </tile>
</tiles>

<layout tileable="false">
  <fixed_layout name="4x4" width="6" height="6">
    <perimeter type="io" priority="100"/>
    <corners type="EMPTY" priority="101"/>
    <fill type="clb" priority="10"/>
    <col type="clb_mul16x16" startx="3" starty="1" priority="20"/>
  </fixed_layout>
</layout>

...

<complexblocklist>
  ...
  <pb_type name="clb_mul16x16">
    <input name="in1" num_pins="2" equivalent="none"/>
    <input name="in2" num_pins="2" equivalent="none"/>
    <output name="out" num_pins="4" equivalent="none"/>
    <clock name="clk" num_pins="1"/>
    <pb_type name="mul16x16" blif_model=".subckt mul16x16" num_pb="1">
      <input name="in1" num_pins="2"/>
      <input name="in2" num_pins="2"/>
      <output name="out" num_pins="4"/>
      <delay_constant max="0.3e-9" in_port="mul16x16.in1" out_port="mul16x16.out"/>
      <delay_constant max="0.3e-9" in_port="mul16x16.in2" out_port="mul16x16.out"/>
    </pb_type>
    <interconnect>
      <direct name="direct1" input="clb_mul16x16.in1" output="mul16x16.in1"/>
      <direct name="direct2" input="clb_mul16x16.in2" output="mul16x16.in2"/>
      <direct name="direct3" input="mul16x16.out" output="clb_mul16x16.out"/>
    </interconnect>
  </pb_type>
</complexblocklist>
...
```
##### OpenFPGA xml 
```
...
<circuit_library>
  ...
  <circuit_model type="hard_logic" name="mul16x16" prefix="mul16x16" is_default="false" verilog_netlist="[path]/mul16x16.v">
    <design_technology type="cmos"/>
    <input_buffer exist="false"/>
    <output_buffer exist="false"/>
    <port type="input" prefix="in1" lib_name="in1" size="2"/>
    <port type="input" prefix="in2" lib_name="in2" size="2"/>
    <port type="output" prefix="out" lib_name="out" size="4"/>
  </circuit_model>
</circuit_library>
...
<pb_type_annotations>
  ...
  <pb_type name="clb_mul16x16.mul16x16" circuit_model_name="mul16x16"/>
</pb_type_annotations>
```
where `[path]` is just a shortcut here and `mul16x16.v` is the following:
```
module mul16x16 (in1, in2, out);
  input  [1:0] in1; 
  input  [1:0] in2;
  output [3:0] out;

  assign out = in1*in2;
endmodule
```
My test module is a SV module which tries to implement a registered multiplier:
```
module custom_mul_test
  (
    input  logic A0, A1, B0, B1,
    input  logic clk,
    output logic P0, P1, P2, P3	
  );

  always @(posedge clk) begin
    {P3, P2, P1, P0} <= {A1, A0} * {B1, B0};
  end
endmodule
```
I was able to make YOSYS perform the techmap of the multiplier to my custom block by specifying a blackbox for it with a custom techmap verilog file which binds the `*` operator to a `mul16x16` module which then is associated to the following blif model:
```
.model mul16x16
.inputs in1 in2
.outputs out
.blackbox
.end
```
```
=== custom_mul_test ===

   Number of wires:                 16
   Number of wire bits:             16
   Number of public wires:          16
   Number of public wire bits:      16
   Number of memories:               0
   Number of memory bits:            0
   Number of processes:              0
   Number of cells:                  5
     $dff                            4
     mul16x16                        1
```
and the ABC output blif file includes a `.subckt mul16x16` which should be recognized by VPR during P&R:
```
# Benchmark "custom_mul_test" written by ABC on Tue May 17 11:20:35 2022
.model custom_mul_test
.inputs A0 A1 B0 B1 clk
.outputs P0 P1 P2 P3

.latch n17 P0 re clk 2
.latch n21 P1 re clk 2
.latch n25 P2 re clk 2
.latch n29 P3 re clk 2

.subckt mul16x16 in1=n42abc in2=n45abc out=n48abc

.names n42abc
 0
.names n45abc
 0
.names n17
 0
.names n21
 0
.names n25
 0
.names n29
 0
.end

.model mul16x16
.inputs in1 in2
.outputs out
.blackbox
.end
```
but when I launch the vpr tool, the routing does not include the usage of my custom multiplier instead of the classic CLB tiles:
[image]
The red square shows the column of my custom multipliers I meant to use but the VPR place and rout does not use them.
Can somebody suggest me a solution or point me to the right direction by telling me what I am doing wrong? 
Thank you so much

tangxifan, 2022-06-01:
@cristianomodesti Can you check in your openfpgashell.log? There are device utilization stats?

cristianomodesti, 2022-06-04:
> @cristianomodesti Can you check in your openfpgashell.log? There are device utilization stats?

Actually no device utilisation. But I found out that if I split the ports on the verilog description of the custom block `mul16x16`, the tool correctly recognises it and correctly maps it.