lnis-uofu/OpenFPGA #667: OPENFPGA SHELL RUN failed with binding a standard inverter cell

AyaseErii, 2022-05-24:
> **Describe the bug**
In the openfpga flow, I am trying to bind an inverter from sky130 standard cell library to one of the openfpga arch files, it shows two errors after executing "run_fpga_task" at full testbench:
Error 1: Miss a default circuit model for the type inv_buf! Try to define it in your architecture file!
Error 2: Command 'read_openfpga_arch' execution has fatal errors
>However, if I bind an OR2 gate only from the same cell library to the same architecture following the tutorial, I can run everything in full testbench successfully once I changed the "iverilog_output.txt" accordingly.

> **To Reproduce**
> Steps to reproduce the behavior:
> 1. Changing the circuit model as the following:

    <circuit_model type="inv_buf" name="sky130_fd_sc_lp__inv_1" prefix="sky130_fd_sc_lp__inv_1" is_default="false" verilog_netlist="{$OPENFPGA_PATH}/skywater-pdk/libraries/sky130_fd_sc_lp/cells/inv/sky130_fd_sc_lp__inv_1.v
       <design_technology type="cmos" topology="inverter" size="1"/>
       <port type="input" prefix="in" lib_name="A" size="1"/>
       <port type="output" prefix="out" lib_name="Y" size="1"/>
    </circuit_model>

>2. Replacing "INVTX1" with "sky130_fd_sc_lp__inv_1" in the whole arch file.
>3. Run 
    
    python3 openfpga_flow/scripts/run_fpga_task.py openfpga_flow/tasks/basic_tests/full_testbench/configuration_chain --debug --show_thread_logs

>Then, you will see two errors:
> Error 1: Miss a default circuit model for the type inv_buf! Try to define it in your architecture file!
> Error 2: Command 'read_openfpga_arch' execution has fatal errors

> **Expected behavior**
I think binding an inverter should be similar to binding an OR2 gate.

> **Enviornment (please complete the following information):**

>  - OS: 
Ubuntu 18.04
>  - Compiler:
gcc-7
clang-6
> - Version:
Current master

> **Additional context**
I took a look at Standard-cell flow #225, this cannot help with solving the problem.
Is there anyone who can help me out with this issue and point me to the right direction? I will appreciate it.

tangxifan, 2022-06-01:
@AyaseErii This is because of the missing syntax ``is_default`` for a given type of circuit models.

https://openfpga.readthedocs.io/en/master/manual/arch_lang/circuit_library/#cmdoption-arg-circuit_model

AyaseErii, 2022-06-01:
Thanks! It works when I add ` is_default=true`.

AyaseErii, 2022-06-01:
> @AyaseErii This is because of the missing syntax `is_default` for a given type of circuit models.
> 
> https://openfpga.readthedocs.io/en/master/manual/arch_lang/circuit_library/#cmdoption-arg-circuit_model

closed 2022-06-01 by AyaseErii