lnis-uofu/OpenFPGA #670: Optimize bitstream generation

alaindargelas, 2022-05-25:
@ManadherRS, please follow up, this PR only partially solves the problem:
https://github.com/lnis-uofu/OpenFPGA/pull/669#event-6677703166

tangxifan, 2022-05-26:
Hi @alaindargelas Thanks for logging this.

@ManadherRS I would like to share my thoughts on further optimization.

- Currently, I use encoded data in the storage of bitstream data structure, which can efficiently reduce the peak memory usage. However, encoding and decoding of the data could be computing-intensive, leading to long runtime. We already see in the CI tests that bitstream generation regression tests are now ~40 minutes. 

Therefore, I think runtime optimization is one of our focus (beyond the current limitation I have stated)

- First of all, I think we should go with the encoding of bitstream raw data when storing them in the data structure. However, we should consider to avoid redundant decoding-encoding. 

https://github.com/lnis-uofu/OpenFPGA/blob/70e2330f1740786ac7a108fd4182d63ab75069d3/openfpga/src/fpga_bitstream/build_fabric_bitstream_memory_bank.cpp#L180-L203

You see that we first decode a number into 1 hot code, and then encode the data again in data structure, 

https://github.com/lnis-uofu/OpenFPGA/blob/70e2330f1740786ac7a108fd4182d63ab75069d3/openfpga/src/fpga_bitstream/fabric_bitstream.cpp#L156-L157

This has wasted a lot of computing time, in my view.
If we can avoid the two-step decode-encoding (for example, find a way to store the address number and only decode when needed), we should be able to improve a lot.

In addition, the bitstream data structure is a common data structure used by different configuration protocols. Please check the codes in 

https://github.com/lnis-uofu/OpenFPGA/blob/master/openfpga/src/fpga_bitstream/build_fabric_bitstream.cpp
https://github.com/lnis-uofu/OpenFPGA/blob/master/openfpga/src/fpga_bitstream/build_fabric_bitstream_memory_bank.cpp

This may help in optimizing the data structure without a strong bias.

tangxifan, 2022-09-06:
Applied a quick patch in https://github.com/lnis-uofu/OpenFPGA/pull/760

closed 2022-09-06 by tangxifan