lnis-uofu/OpenFPGA #672: Preconfigured testbench not working but full_testbench does

mattvlsi, 2022-05-26:
I have used benchmark_sweep/counter8 task with k6_frac_N10_tileable_40nm architecture. Full testbench successfully runs and give expected results. However, within same task I also generate preconfigured testbench but it does not give expected results. GPIO ports give 1'bx results. What would be my problem? How can I solve this issue? I attached the task file as in form of directly ready-to-run. 

Reproduce:
-Extract the file and run the command of "run-task"
[preconf_testbench.zip](https://github.com/lnis-uofu/OpenFPGA/files/8779118/preconf_testbench.zip)

mattvlsi, 2022-06-03:
Hi @tangxifan. I used counter8 testcase. But openfpga arch is tileable k6n10 without dpram and DSP blocks. The attached file is a ready-to-run task repo.

tangxifan, 2022-06-04:
@mattvlsi I do not see your attachment. Would you mind to re-upload? Thanks

mattvlsi, 2022-06-06:
Here is my task folder:
[preconf_testbench.zip](https://github.com/lnis-uofu/OpenFPGA/files/8844551/preconf_testbench.zip)

tangxifan, 2022-06-26:
@ghost I see in the openfpgashell.log that OpenFPGA runs without errors.
I believe the problem comes from the openfpgashell script which writes both full testbench and preconfigured testbench. This will cause some errors when running HDL simulations using OpenFPGA flow scripts.
Can you try to comment out the LINE 58?

https://github.com/lnis-uofu/OpenFPGA/blob/ef264e59cef36f3ade048a26f46a4feae6b7d487/openfpga_flow/openfpga_shell_scripts/example_without_ace_script.openfpga#L52-L60

msaideroglu, 2022-09-14:
Hi @tangxifan. I am facing the same issue described above. I um running **/openfpga_flow/tasks/fpga_verilog/bram/dpram1k** task. It successfully completes as displayed in shell. However preconfigured testbench can not display true results. All Ram input and outputs are displaying 1'bx as well as behavioral benchmark ports. However if I run full_testbench, benchmark ports displays 1'bx again but fpga ports display logic values, but I can not check if the design is correctly placed and routed. I also tried to comment out LINE 58 full_testbench line to get true results for preconfigured testbench, result was same as before. I attached preconf and full testbench views:
[image]
(preconfigured tb results with 1'bx's)
[image]
(full tb results with wrong benchmark results)

**To reproduce:**
run "run-task dpram1k" in the task folder