lnis-uofu/OpenFPGA #683: Layout generation of a custom design

AyaseErii, 2022-06-12:
**Is your feature request related to a problem? Please describe.**
I tried to use OpenFPGA to generate a custom FPGA fabric. From the generated FPGA fabric Verilog files, I found that the fabric_top.v describes one tile rather than n x n tile array. After I generated the GDSII from the generated FPGA fabric, it is just a single tile, not an array. So my questions are: Is it the only way to use Cadence Innovus to do a duplication of that single tile for placing and routing a n x n array? Is there anyway (or feature) to specify the layout scale (say 8x8 or more) in OpenFPGA .xml and vpr_arch file, and then to auto-generate an array-like FPGA fabric for GDSII generation?

Thank you in advance!

tangxifan, 2022-06-13:
@AyaseErii OpenFPGA allows you to define any array size thru VPR's architecture description language.

See details in the example: 

https://github.com/lnis-uofu/OpenFPGA/blob/d2975071dd7e408ca799e4e430ae55d17ce9fcf2/openfpga_flow/tasks/basic_tests/fix_pins/config/task.conf#L22

AyaseErii, 2022-06-13:
Thanks for the help. I know what to do to generate larger fabric now.

tangxifan, 2022-09-06:
Close the issue as answered. If you need further help, please comment (reopen this issue) or create a new issue.
Thank you for the interests in OpenFPGA.

closed 2022-09-06 by tangxifan