lnis-uofu/OpenFPGA #684: VTR gives error when using bigger architectures

yunuseryilmaz18, 2022-06-14:
> **Describe the bug**
  I added a new fixed layout configuration to the k4_N4_tileable_40nm VPR architecture description file but VPR gives an error in the placement stage:
  ```
  /mnt/HDD/OpenFPGA/vpr/src/route/rr_graph2.cpp:1377 get_rr_node_index: Assertion 'ptc >= 0' failed.
  Aborted (core dumped)
  ```
  The width and height of my layout is 8. Also, I tried 5x5, 6x6, and 7x7 and they worked without errors.

> **Which part of OpenFPGA is buggy**
> [X] VPR

> **To Reproduce**
> Steps to reproduce the behavior:
> 1. Clone OpenFPGA repository and checkout commit id: d2975071dd7e408ca799e4e430ae55d17ce9fcf2
> 2. Open k4_N4_tileable_40nm.xml and add these lines to `<layout>` section:
```
<fixed_layout name="3x3" width="8" height="8">
      <!--Perimeter of 'io' blocks with 'EMPTY' blocks at corners-->
      <perimeter type="io" priority="100"/>
      <corners type="EMPTY" priority="101"/>
      <!--Fill with 'clb'-->
      <fill type="clb" priority="10"/>
    </fixed_layout>
```
> 3. Run VPR with `./vpr k4_N4_tileable_40nm.xml counter.blif --disp on --device 3x3` in OpenFPGA/vpr directory.
> 4. See the error.

> **Expected behavior**
> I want to see the placement and routing of my circuit on the 8x8 layout.

> **Screenshots**
> [image]

> **Enviornment (please complete the following information):**

>  - OS: Ubuntu 20.04

>  - Compiler: GCC 9.4.0

> - Version: Current master

> **Additional context**
> I added counter.blif file as a zip file below:
[counter.zip](https://github.com/lnis-uofu/OpenFPGA/files/8897240/counter.zip)

tangxifan, 2022-06-26:
@yunuseryilmaz18 This looks weird. I see the placer is building a non-tileable routing resource graph.
Can you try to use a fixed routing channel width? For example, ``--route_chan_width 100``
See if it can run or not. If still not, it should be bug like 

https://github.com/lnis-uofu/OpenFPGA/issues/71

yunuseryilmaz18, 2022-06-29:
Yes, it works now. Thank you. I'm closing the issue.

closed 2022-06-29 by yunuseryilmaz18