lnis-uofu/OpenFPGA #687: How to generate an island-style FPGA array layout

AyaseErii, 2022-06-17:
**Is your feature request related to a problem? Please describe.**
I generated a 8x8 FPGA fabric by the OpenFPGA, and then tried to generate a corresponding GDSII by an ASIC tool. I expected to see the 8x8 clearly on the final GDSII. However, the actual generated GDSII has just a relatively large area fulfilled with uniform-distributed wires and cells, i.e. not 8x8 island-style array at all.

Here is the layout of the so-called 8x8 FPGA:
[image]

However, this is a 20x20 FPGA layout from OpenFPGA FPL'19 paper, which is island-style:
[image]

Here is a quote from the paper:

"We developed a floorplanning script to place CLBs in an island-style with sufficient area margin for CBs and SBs."

Therefore, I would like to know if there is a feature in OpenFPGA to help with generating an island-style layout? If yes, how can I use that feature? Could you share the floorplanning script as a reference? I will appreciate it! 

Thanks!

AyaseErii, 2022-06-17:
@ganeshgore Thanks for your sharing! I will take a look at this paper.

AyaseErii, 2022-06-22:
@ganeshgore Hi Ganesh, thanks again for your help, after reading the paper, I got a general idea about hierarchical floorplanning. I looked at the fabric generated by OpenFPGA. In SRC dictionary, there are several sub-dictionaries: `lb`, `routing`, and `submodule` as you said in the hierarchy. 

However, when I tried to compare my generated fabric with the SOFA project fabric, I happened to find something like `fpga_core.v` was included in the `fabric_netlist.v` in the SOFA, while there is just `fabric_top.v` was included in the ``fabric_netlist.v`` in my fabric. In addition to that, `fabric_top.v` in the `sub_module` as well. I am kind of confused here. 

So my questions are:

1. Does OpenFPGA will automatically generate `fpga_core.v`? If yes, how can I generated it? 

2. How can I generate `fabric_top.v` in the `sub_module` with OpenFPGA? 

I believe I can understand more about the preparation with OpenFPGA for the hierarchical floor planning if I can get these concepts/files clearly. Thanks!

yunuseryilmaz18, 2022-06-29:
@ganeshgore Hello, is it possible to share PnR scripts for OpenLane or another open-source EDA tool? Do you have any?

ganeshgore, 2022-06-29:
@AyaseErii

The fpga_top generated using OpenFPGA has been rename to fpga_core during restructuring, and new fpga_top is just one to one mapping to the fpga_core.

We are in process of developing more sophisticate flow to perform the netlist restructuring, you can expect public release in the coming month.

AyaseErii, 2022-06-30:
@ganeshgore Thank you very much, Ganesh! I think now I understand the differences between fpga_top and fpga_core.

tangxifan, 2022-09-06:
Close the issue as answered. If you need further help, please comment (reopen this issue) or create a new issue.
Thank you for the interests in OpenFPGA.

closed 2022-09-06 by tangxifan

zaidtahirbutt, 2023-05-06:
> Hello AyaseErii, You need to perform hierarchical floorplanning to achieve a tileable layout as detailed in the 20x20 FPGA layout. The OpenFPGA-generated Verilog netlist is already hierarchical, which saves you time to create tiles yourself. Refer to this for the hierarchy https://openfpga.readthedocs.io/en/latest/manual/fpga_verilog/fabric_netlist/
> 
> You can use these tiles with any commercial toolchain and hierarchical design flow. Unfortunately, the script used for this and following takeouts is not open source (as it uses a commercial toolchain), but you can refer following paper for more detail on the hierarchical physical design using an OpenFPGA-generated netlist.
> 
> https://drive.google.com/file/d/1THHJZbfBKg-Y4V7UpJ1PScLXSUwcthp9/view

Thanks for the paper. Helped to understand better.

Lukemagik, 2024-04-19:
Hi @ganeshgore , have you managed to create a flow for an open-source place and route tool?
For example OpenLane