lnis-uofu/OpenFPGA #704: Some questions about Memory Bank Protocol

mattvlsi, 2022-07-04:
Hi. I'm trying to do PnR of an OpenFPGA. My technology node is 250nm. So DFF chain based architectures cost a huge area because of DFF's. To be able to reduce area, I saw memory bank based architectures. I also saw that ql_memory_bank protocol is physical design friendly. Should I continue with ql_memory_bank protocol architectures for physical design? Does this choice give me area cost gain that I expected? What is its disadvantages? Further, there are only k4n4 architecture files that support memory bank protocol. I want to do a design of k4n8 or k6n10. How can I configure k4n8 architecture for my purpose? @tangxifan @ganeshgore

ganeshgore, 2022-07-04:
Hello @mattvlsi 
`ql_memory_bank` protocol used latches instead of flip-flops. 
Definitely, there is an area benefit but it depends on how the latch cell is implemented in the standard cell library. 

**Challenge of using memory bank**:
 `ql_memory_bank` protocol is essentially like SRAM addressing, which has a long bit and word line running across the fabric.
It shud be straightforward if you are performing flat place and route.
 In the case of hierarchical/tileable flow, the physical design requires a lot more effort (as OpenFPGA does not generate feedthrough for these lines), also based on the place and route tool you use there could be high routing congestion on the boundary interface of each tile.

**Configuring K4N8 or K6N10** 
You can configure any architecture to use memory bank protocol, makes sure you set `<organization type="ql_memory_bank"`
and define `circuit_model type="sram"` with `WL` and `BL` lines defined, Alternately you can copy from the existing OpenFPGA architecture file.

tangxifan, 2022-07-04:
@mattvlsi Please note that ``ql_memory_bank`` will replace ``memory_bank`` in future. Current ``memory_bank`` will be deprecated.
Therefore, I suggest to start with ``ql_memory_bank`` to avoid any backward compatibility issues.

msaideroglu, 2022-08-23:
Thanks a lot for your detailed explanations.

closed 2022-08-23 by msaideroglu