lnis-uofu/OpenFPGA #705: Failed to find matching architecture model for 'ALU'

yunuseryilmaz18, 2022-07-05:
Hello, I want to benchmark FPGA architectures with a RISC-V processor. I've created blif file of the processor and run VPR with `./vpr k4_frac_N4_tileable_40nm.xml riscv-cpu.blif --disp on` but it gives `Failed to find matching architecture model for 'ALU'` error. How can I run designs with multiple files in VPR?

tangxifan, 2022-07-07:
@yunuseryilmaz18 It looks like you have defined a blackbox ``ALU`` in Yosys when synthesis.

[image]

Please double check your synthesis scripts. If you are using OpenFPGA's example scripts, you expect to see only LUTs and FFs.

https://github.com/lnis-uofu/OpenFPGA/blob/master/openfpga_flow/misc/ys_tmpl_yosys_vpr_flow.ys

Currently, VPR supports single BLIF and eBLIF.

yunuseryilmaz18, 2022-07-25:
I've added -noblackbox argument to read_verilog command and it works now. Thank you.

closed 2022-07-25 by yunuseryilmaz18