lnis-uofu/OpenFPGA #706: Synthesis of RISC processor does not end

yunuseryilmaz18, 2022-07-07:
> **Describe the bug**
> I created a template task and ran fabric generation with `RISC_posedge_clk`, located in the micro_benchmark folder, as a benchmark circuit. I've waited to see the results for 7+ hours, but Yosys never finishes the synthesis. It stucks in the `proc` command and the system finally kills the process.

> **Which part of OpenFPGA is buggy**
> [X] Yosys

> **To Reproduce**
> Steps to reproduce the behavior: 
> 1. Clone OpenFPGA repository and checkout commit id: 37ecb6ca4c922fd41c22c40a19937613632fdaf2
> 2. Create a yosys_vpr task with `create-task project1 yosys_vpr` command
> 3. Change benchmark circuit and top module for `RISC_posedge_clk`.
> 4. Execute `run-task project1`
> 5. See the error.

> **Expected behaviour**
> I want to test a RISC processor on a k6_frac_N10 architecture.

> **Screenshots**
> [image]

> **Enviornment (please complete the following information):**

>  - OS: 
> - [X] Ubuntu 20.04

>  - Compiler:
> - [X] gcc-9

> - Version:
> - [X] Current master

> **Additional context**
> Add any other context about the problem here.

tangxifan, 2022-07-12:
@yunuseryilmaz18 It is hard to diagnose without details. The benchmark ``RISC_posedge_clk`` is covered in regression tests

https://github.com/lnis-uofu/OpenFPGA/blob/master/openfpga_flow/tasks/fpga_bitstream/generate_bitstream/configuration_chain/device_96x96/config/task.conf

Yosys does finish in that case. Maybe it is better that you can share your yosys scripts.

yunuseryilmaz18, 2022-07-25:
I've run the task you mentioned and the synthesis finish when I run it. The mistake I made was the top module, I set the top module as RISC_core_mem_top. After changing the top module to RISC_core_top, the synthesis finished successfully.

closed 2022-07-25 by yunuseryilmaz18