lnis-uofu/OpenFPGA #710: Simple questions about OpenFPGA

Baungarten-CINVESTAV, 2022-07-11:
Hello every one,

I’m trying to implement an FPGA capable of running the AES128 algorithm. Trying to develop that FPGA some problems/questions appear.

1-	How can I develop my own operating testbench, when I implement the Operating Phase testbench the signals that it creates are random, so the reset, load or decrypt signal are on and off, making it impossible for the algorithm to finish. 
(I use the “preconfigured_testbench_without_self_checking_example_script” shell template)
 
[image]

2-	Why the OpenFPGA flow consumes a lot of RAM?, e.g. to run the avobe example I use the k6_frac_N10_tileable_40nm architecture, and 24 by 24 FPGA (width="24" height="24"), it consumes 95% of my RAM (118 GB), why this happened and how I can avoid it? 
 
[image]

3-	As the above image shows that took a lot of time, if I want to run another testbench, how can I reduce the implementation time? Or what shell template can I run for just to load my design and check the testbench?

I hope someone could help me.

tangxifan, 2022-07-12:
@Baungarten-CINVESTAV Can you share the openfpga XML file? It matters which configuration protocol you are using.

Baungarten-CINVESTAV, 2022-07-12:
> 
[arch.zip](https://github.com/lnis-uofu/OpenFPGA/files/9096096/arch.zip)

tangxifan, 2022-07-12:
@Baungarten-CINVESTAV This is not normal when using the configuration chain protocol.  Can you share your openfpgashell.log and task.conf? 
I can check where the memory leaks happens.

Baungarten-CINVESTAV, 2022-07-12:
> @Baungarten-CINVESTAV This is not normal when using the configuration chain protocol. Can you share your openfpgashell.log and task.conf? I can check where the memory leaks happens.

yes, here are the files

[Shell_config.zip](https://github.com/lnis-uofu/OpenFPGA/files/9097118/Shell_config.zip)

tangxifan, 2022-07-12:
@Baungarten-CINVESTAV I double checked your log files. OpenFPGA's peak memory usage is 242MB. 

```
Generating SDC for Timing/Power analysis on the mapped FPGA './SDC_analysis/aes_fpga_top_analysis.sdc' took 2.08 seconds (max_rss 242.4 MiB, delta_rss +0.0 MiB)

Command line to execute: exit

Confirm selected options when call command 'exit':

Finish execution with 0 errors

The entire OpenFPGA flow took 81.8226 seconds

Thank you for using OpenFPGA!
```

You see a huge memory usage when running iVerilog simulation. This is expected. For large FPGA designs, I suggest to use commercial simulators, e.g., Modelsim, Ncsim and VCS.

closed 2022-07-14 by Baungarten-CINVESTAV