lnis-uofu/OpenFPGA #718: FPGA Netlist changes when benchmark changes

mustafaarslan0, 2022-07-20:
Hi. I try to generate fabric dependent bitstream to FPGA that generated by OpenFPGA flow. But when I changed the benchmark, OpenFPGA flow generates different FPGA netlist (minor differences) with same architecture files. Why does FPGA netlist change with different benchmark with same architecture files? How can I generate bitstream to exactly same FPGA?

Example differences from file: 
/home/arslan/OpenFPGA/mytask03/run007/k6_frac_N10_tileable_40nm/downcounter_8bit/MIN_ROUTE_CHAN_WIDTH/SRC/routing/sb_0__0_.v 

[image]

Script, config, archs and benchmarks that I used:
[mytask03.zip](https://github.com/lnis-uofu/OpenFPGA/files/9149431/mytask03.zip)

Thanks.

tangxifan, 2022-07-20:
@mustafaarslan0 This is because VPR was run to find the best routable channel width. You can fix the route chan width by referring to the test case: 

https://github.com/lnis-uofu/OpenFPGA/blob/9dca626a3d7161c087d48eb8da5f7fad891b655b/openfpga_flow/tasks/fpga_bitstream/generate_bitstream/configuration_chain/device_96x96/config/task.conf#L23-L24

mustafaarslan0, 2022-07-22:
It works, thanks.

closed 2022-07-22 by mustafaarslan0