lnis-uofu/OpenFPGA #721: Targeting an operating frequency

mattvlsi, 2022-07-21:
I have a fabric that successfully runs benchmarks on it with a 2x2 sized layout. I want to do its pnr works and see how much operating frequency I can achieve with it. What strategy I should follow during applying timing constraints in DC synthesis part? My idea is apply same clk period constraint for all CBs and CLB blocks and make macro with them. So placement can be done. Is it a true approach? Then for instance I applied 25ns period for all of them and passed timing analysis one by one. Then if I place and route those macros, can I say that my FPGA operates with a 25ns clk period? As a summary I have a target frequency that I want to reach, how can I achieve that frequency? Thanks @ganeshgore , @tangxifan .

ganeshgore, 2022-07-21:
​Hello 

The ​​operating frequency of the counter implemented on the FPGA depends upon various factors, like:
(1) Architecture, whether the counter will be implemented on the LUTs, soft logic, or a hard logic 
(2) Synthesis script used in your FPGA CAD flow (if it can infer counter logic and implement on any of the above-specialized structures); in the typical case it will just implemented with the LUT 
(3) and finally your fabric timing delays on the critical path

Coming back to your question about fabric synthesis in the design compiler, Yes, you can certainly apply the same clock constraints to all the blocks, but there there are no data clocks in routing modules (it just has a programming clock that deals with just configuration flipflop) in that case you need to use constraints like `set_max_delays` or create a virtual clock to constraint combinational paths.

25ns constraint is each block does not ensure operating frequency to be 1/25ns. Depending upon your design, your critical path could be between different blocks. In which case, you need to add those delays to get your operating frequency.

My general suggestion for synthesis to keep simple is to constraint different blocks of the FPGA (like CBX, CBY, and CLB) independently with specific timing, area, and power constraints. and estimate operating frequency for the specific benchmark. 

Hope that helps

mattvlsi, 2022-07-21:
Thanks @ganeshgore. I understood it

closed 2022-07-21 by mattvlsi