lnis-uofu/OpenFPGA #724: I'am not sure where does error come from.

omarghoneim3, 2022-07-22:
python3 openfpga_flow/scripts/run_fpga_task.py compilation_verification
 INFO (     MainThread) - Set up to run 2 Parallel threads
 INFO (     MainThread) - Currently running task compilation_verification
 INFO (     MainThread) - Created "run004" directory for current task run
 INFO (     MainThread) - Running "vpr_blif" flow
 INFO (     MainThread) - Found 1 Architectures 1 Benchmarks & 1 Script Parameters
 INFO (     MainThread) - Created total 1 jobs
ERROR (00_and2_MIN_ROUTE_CHAN_WIDTH) - Failed to execute openfpga flow - 00_and2_MIN_ROUTE_CHAN_WIDTH
Traceback (most recent call last):
  File "openfpga_flow/scripts/run_fpga_task.py", line 513, in run_single_script
subprocess.CalledProcessError: Command 'python3 /home/omar/OpenFPGA/openfpga_flow/scripts/run_fpga_flow.py /home/omar/OpenFPGA/openfpga_flow/vpr_arch/k4_N4_tileable_40nm.xml /home/omar/OpenFPGA/openfpga_flow/benchmarks/micro_benchmark/and2/and2.blif --top_module and2 --run_dir /home/omar/OpenFPGA/openfpga_flow/tasks/compilation_verification/run004/k4_N4_tileable_40nm/and2/MIN_ROUTE_CHAN_WIDTH --fpga_flow vpr_blif --openfpga_shell_template /home/omar/OpenFPGA/openfpga_flow/openfpga_shell_scripts/example_script.openfpga --openfpga_arch_file /home/omar/OpenFPGA/openfpga_flow/openfpga_arch/k4_N4_40nm_frame_openfpga.xml --openfpga_sim_setting_file /home/omar/OpenFPGA/openfpga_flow/openfpga_simulation_settings/auto_sim_openfpga.xml --activity_file /home/omar/OpenFPGA/openfpga_flow/benchmarks/micro_benchmark/and2/and2.act --base_verilog /home/omar/OpenFPGA/openfpga_flow/benchmarks/micro_benchmark/and2/and2.v --power --power_tech /home/omar/OpenFPGA/openfpga_flow/tech/PTM_45nm/45nm.xml --vpr_fpga_verilog --vpr_fpga_verilog_dir /home/omar/OpenFPGA/openfpga_flow/tasks/compilation_verification/run004/k4_N4_tileable_40nm/and2/MIN_ROUTE_CHAN_WIDTH --vpr_fpga_x2p_rename_illegal_port --end_flow_with_test --vpr_fpga_verilog_formal_verification_top_netlist --TOP and2 --ACT /home/omar/OpenFPGA/openfpga_flow/benchmarks/micro_benchmark/and2/and2.act --VERILOG /home/omar/OpenFPGA/openfpga_flow/benchmarks/micro_benchmark/and2/and2.v --CHAN_WIDTH 300' returned non-zero exit status 0.

omarghoneim3, 2022-07-22:
I tried to debug it, but it doesnot get clearer
 run-task basic_tests/full_testbench/configuration_chain
 INFO (     MainThread) - Set up to run 2 Parallel threads
 INFO (     MainThread) - Currently running task basic_tests/full_testbench/configuration_chain
 INFO (     MainThread) - Created "run002" directory for current task run
 INFO (     MainThread) - Running "yosys_vpr" flow
 INFO (     MainThread) - Found 1 Architectures 3 Benchmarks & 1 Script Parameters
 INFO (     MainThread) - Created total 3 jobs
ERROR (00_and2_MIN_ROUTE_CHAN_WIDTH) - Failed to execute openfpga flow - 00_and2_MIN_ROUTE_CHAN_WIDTH
Traceback (most recent call last):
  File "/home/omar/OpenFPGA/openfpga_flow/scripts/run_fpga_task.py", line 513, in run_single_script
    raise subprocess.CalledProcessError(0, " ".join(command))
subprocess.CalledProcessError: Command 'python3 /home/omar/OpenFPGA/openfpga_flow/scripts/run_fpga_flow.py /home/omar/OpenFPGA/openfpga_flow/vpr_arch/k4_N4_tileable_40nm.xml /home/omar/OpenFPGA/openfpga_flow/benchmarks/micro_benchmark/and2/and2.v --top_module and2 --run_dir /home/omar/OpenFPGA/openfpga_flow/tasks/basic_tests/full_testbench/configuration_chain/run002/k4_N4_tileable_40nm/and2/MIN_ROUTE_CHAN_WIDTH --fpga_flow yosys_vpr --openfpga_shell_template /home/omar/OpenFPGA/openfpga_flow/openfpga_shell_scripts/write_full_testbench_example_script.openfpga --openfpga_arch_file /home/omar/OpenFPGA/openfpga_flow/openfpga_arch/k4_N4_40nm_cc_openfpga.xml --openfpga_sim_setting_file /home/omar/OpenFPGA/openfpga_flow/openfpga_simulation_settings/auto_sim_openfpga.xml --openfpga_vpr_device_layout  --openfpga_fast_configuration  --power --power_tech /home/omar/OpenFPGA/openfpga_flow/tech/PTM_45nm/45nm.xml --vpr_fpga_verilog --vpr_fpga_verilog_dir /home/omar/OpenFPGA/openfpga_flow/tasks/basic_tests/full_testbench/configuration_chain/run002/k4_N4_tileable_40nm/and2/MIN_ROUTE_CHAN_WIDTH --vpr_fpga_x2p_rename_illegal_port --end_flow_with_test --TOP and2 --CHAN_WIDTH 300 --READ_VERILOG_OPTIONS -nolatches' returned non-zero exit status 0.

tangxifan, 2022-07-22:
@omarghoneim3 These options will help you on getting more debugging messages

https://openfpga.readthedocs.io/en/master/dev_manual/regression_tests/#test-options

omarghoneim3, 2022-07-22:
I try to debug, but it shows me this
python3 openfpga_flow/scripts/run_fpga_task.py compilation_verification--debug
 INFO (     MainThread) - Set up to run 2 Parallel threads
 INFO (     MainThread) - Currently running task compilation_verification--debug
ERROR (     MainThread) - Task directory [['compilation_verification--debug']] not found locally at [compilation_verification--debug], absolutely at [/compilation_verification--debug], or in OpenFPGA task directory [/home/omar/OpenFPGA/openfpga_flow/tasks/compilation_verification--debug]
ERROR (     MainThread) - Exiting . . . . . .

omarghoneim3, 2022-07-22:
python3 openfpga_flow/scripts/run_fpga_task.py compilation_verification--show_thread_logs
 INFO (     MainThread) - Set up to run 2 Parallel threads
 INFO (     MainThread) - Currently running task compilation_verification--show_thread_logs
ERROR (     MainThread) - Task directory [['compilation_verification--show_thread_logs']] not found locally at [compilation_verification--show_thread_logs], absolutely at [/compilation_verification--show_thread_logs], or in OpenFPGA task directory [/home/omar/OpenFPGA/openfpga_flow/tasks/compilation_verification--show_thread_logs]
ERROR (     MainThread) - Exiting . . . . . .

tangxifan, 2022-07-22:
@omarghoneim3  I think you miss the space between compilation_verification--debug

omarghoneim3, 2022-07-22:
python3 openfpga_flow/scripts/run_fpga_task.py compilation_verification --debug
 INFO (     MainThread) - Setting loggger in debug mode
 INFO (     MainThread) - Set up to run 2 Parallel threads
 INFO (     MainThread) - Currently running task compilation_verification
 INFO (     MainThread) - Created "run007" directory for current task run
 INFO (     MainThread) - Running "vpr_blif" flow
 INFO (     MainThread) - Found 1 Architectures 1 Benchmarks & 1 Script Parameters
 INFO (     MainThread) - Created total 1 jobs
DEBUG (00_and2_MIN_ROUTE_CHAN_WIDTH) - Running OpenFPGA flow with [['python3', '/home/omar/OpenFPGA/openfpga_flow/scripts/run_fpga_flow.py', '/home/omar/OpenFPGA/openfpga_flow/vpr_arch/k4_N4_tileable_40nm.xml', '/home/omar/OpenFPGA/openfpga_flow/benchmarks/micro_benchmark/and2/and2.blif', '--top_module', 'and2', '--run_dir', '/home/omar/OpenFPGA/openfpga_flow/tasks/compilation_verification/run007/k4_N4_tileable_40nm/and2/MIN_ROUTE_CHAN_WIDTH', '--fpga_flow', 'vpr_blif', '--openfpga_shell_template', '/home/omar/OpenFPGA/openfpga_flow/openfpga_shell_scripts/example_script.openfpga', '--openfpga_arch_file', '/home/omar/OpenFPGA/openfpga_flow/openfpga_arch/k4_N4_40nm_frame_openfpga.xml', '--openfpga_sim_setting_file', '/home/omar/OpenFPGA/openfpga_flow/openfpga_simulation_settings/auto_sim_openfpga.xml', '--activity_file', '/home/omar/OpenFPGA/openfpga_flow/benchmarks/micro_benchmark/and2/and2.act', '--base_verilog', '/home/omar/OpenFPGA/openfpga_flow/benchmarks/micro_benchmark/and2/and2.v', '--power', '--power_tech', '/home/omar/OpenFPGA/openfpga_flow/tech/PTM_45nm/45nm.xml', '--vpr_fpga_verilog', '--vpr_fpga_verilog_dir', '/home/omar/OpenFPGA/openfpga_flow/tasks/compilation_verification/run007/k4_N4_tileable_40nm/and2/MIN_ROUTE_CHAN_WIDTH', '--vpr_fpga_x2p_rename_illegal_port', '--end_flow_with_test', '--vpr_fpga_verilog_formal_verification_top_netlist', '--debug', '--TOP', 'and2', '--ACT', '/home/omar/OpenFPGA/openfpga_flow/benchmarks/micro_benchmark/and2/and2.act', '--VERILOG', '/home/omar/OpenFPGA/openfpga_flow/benchmarks/micro_benchmark/and2/and2.v', '--CHAN_WIDTH', '300']]
ERROR (00_and2_MIN_ROUTE_CHAN_WIDTH) - Failed to execute openfpga flow - 00_and2_MIN_ROUTE_CHAN_WIDTH
Traceback (most recent call last):
  File "openfpga_flow/scripts/run_fpga_task.py", line 513, in run_single_script
subprocess.CalledProcessError: Command 'python3 /home/omar/OpenFPGA/openfpga_flow/scripts/run_fpga_flow.py /home/omar/OpenFPGA/openfpga_flow/vpr_arch/k4_N4_tileable_40nm.xml /home/omar/OpenFPGA/openfpga_flow/benchmarks/micro_benchmark/and2/and2.blif --top_module and2 --run_dir /home/omar/OpenFPGA/openfpga_flow/tasks/compilation_verification/run007/k4_N4_tileable_40nm/and2/MIN_ROUTE_CHAN_WIDTH --fpga_flow vpr_blif --openfpga_shell_template /home/omar/OpenFPGA/openfpga_flow/openfpga_shell_scripts/example_script.openfpga --openfpga_arch_file /home/omar/OpenFPGA/openfpga_flow/openfpga_arch/k4_N4_40nm_frame_openfpga.xml --openfpga_sim_setting_file /home/omar/OpenFPGA/openfpga_flow/openfpga_simulation_settings/auto_sim_openfpga.xml --activity_file /home/omar/OpenFPGA/openfpga_flow/benchmarks/micro_benchmark/and2/and2.act --base_verilog /home/omar/OpenFPGA/openfpga_flow/benchmarks/micro_benchmark/and2/and2.v --power --power_tech /home/omar/OpenFPGA/openfpga_flow/tech/PTM_45nm/45nm.xml --vpr_fpga_verilog --vpr_fpga_verilog_dir /home/omar/OpenFPGA/openfpga_flow/tasks/compilation_verification/run007/k4_N4_tileable_40nm/and2/MIN_ROUTE_CHAN_WIDTH --vpr_fpga_x2p_rename_illegal_port --end_flow_with_test --vpr_fpga_verilog_formal_verification_top_netlist --debug --TOP and2 --ACT /home/omar/OpenFPGA/openfpga_flow/benchmarks/micro_benchmark/and2/and2.act --VERILOG /home/omar/OpenFPGA/openfpga_flow/benchmarks/micro_benchmark/and2/and2.v --CHAN_WIDTH 300' returned non-zero exit status 0.

tangxifan, 2022-07-22:
Can you add option ``--show_thread_logs``?

omarghoneim3, 2022-07-22:
python3 openfpga_flow/scripts/run_fpga_task.py compilation_verification --show_thread_logs
 INFO (     MainThread) - Set up to run 2 Parallel threads
 INFO (     MainThread) - Currently running task compilation_verification
 INFO (     MainThread) - Created "run008" directory for current task run
 INFO (     MainThread) - Running "vpr_blif" flow
 INFO (     MainThread) - Found 1 Architectures 1 Benchmarks & 1 Script Parameters
 INFO (     MainThread) - Created total 1 jobs
 INFO (00_and2_MIN_ROUTE_CHAN_WIDTH) - Traceback (most recent call last):
 INFO (00_and2_MIN_ROUTE_CHAN_WIDTH) -   File "/home/omar/OpenFPGA/openfpga_flow/scripts/run_fpga_flow.py", line 22, in <module>
 INFO (00_and2_MIN_ROUTE_CHAN_WIDTH) -     from envyaml import EnvYAML
 INFO (00_and2_MIN_ROUTE_CHAN_WIDTH) - ModuleNotFoundError: No module named 'envyaml'
ERROR (00_and2_MIN_ROUTE_CHAN_WIDTH) - Failed to execute openfpga flow - 00_and2_MIN_ROUTE_CHAN_WIDTH
Traceback (most recent call last):
  File "openfpga_flow/scripts/run_fpga_task.py", line 513, in run_single_script
subprocess.CalledProcessError: Command 'python3 /home/omar/OpenFPGA/openfpga_flow/scripts/run_fpga_flow.py /home/omar/OpenFPGA/openfpga_flow/vpr_arch/k4_N4_tileable_40nm.xml /home/omar/OpenFPGA/openfpga_flow/benchmarks/micro_benchmark/and2/and2.blif --top_module and2 --run_dir /home/omar/OpenFPGA/openfpga_flow/tasks/compilation_verification/run008/k4_N4_tileable_40nm/and2/MIN_ROUTE_CHAN_WIDTH --fpga_flow vpr_blif --openfpga_shell_template /home/omar/OpenFPGA/openfpga_flow/openfpga_shell_scripts/example_script.openfpga --openfpga_arch_file /home/omar/OpenFPGA/openfpga_flow/openfpga_arch/k4_N4_40nm_frame_openfpga.xml --openfpga_sim_setting_file /home/omar/OpenFPGA/openfpga_flow/openfpga_simulation_settings/auto_sim_openfpga.xml --activity_file /home/omar/OpenFPGA/openfpga_flow/benchmarks/micro_benchmark/and2/and2.act --base_verilog /home/omar/OpenFPGA/openfpga_flow/benchmarks/micro_benchmark/and2/and2.v --power --power_tech /home/omar/OpenFPGA/openfpga_flow/tech/PTM_45nm/45nm.xml --vpr_fpga_verilog --vpr_fpga_verilog_dir /home/omar/OpenFPGA/openfpga_flow/tasks/compilation_verification/run008/k4_N4_tileable_40nm/and2/MIN_ROUTE_CHAN_WIDTH --vpr_fpga_x2p_rename_illegal_port --end_flow_with_test --vpr_fpga_verilog_formal_verification_top_netlist --TOP and2 --ACT /home/omar/OpenFPGA/openfpga_flow/benchmarks/micro_benchmark/and2/and2.act --VERILOG /home/omar/OpenFPGA/openfpga_flow/benchmarks/micro_benchmark/and2/and2.v --CHAN_WIDTH 300' returned non-zero exit status 0.

GaoGuanxi, 2022-07-22:
ModuleNotFoundError: No module named 'envyaml'

tangxifan, 2022-07-22:
Remember to install python dependencies: (See doc https://openfpga.readthedocs.io/en/master/tutorials/getting_started/compile/)

```
python3 -m pip install -r requirements.txt
```

omarghoneim3, 2022-07-22:
thank you so much :)

closed 2022-07-22 by omarghoneim3