lnis-uofu/OpenFPGA #730: Benchmarking a FIFO circuit

yunuseryilmaz18, 2022-07-26:
> **Describe the bug**
> Hi. I've changed the benchmark circuit in [fpga_verilog/bram/dpram1k](https://github.com/lnis-uofu/OpenFPGA/blob/master/openfpga_flow/tasks/fpga_verilog/bram/dpram1k/config/task.conf) task as [FIFO](https://github.com/lnis-uofu/OpenFPGA/blob/master/openfpga_flow/benchmarks/micro_benchmark/fifo/rtl/fifo.v) circuit. However, Yosys creates black boxes which are not suitable for PnR in VPR, such as $alu and $_DFFE_PP1P_ and this makes OpenFPGA flow crash.

> **Which part of OpenFPGA is buggy**
> - [X] VPR

> **To Reproduce**
> Steps to reproduce the behavior:
> 1. Clone OpenFPGA repository and checkout commit id: 2e8d5aa65e36d3b185725848723d53c8ab9dbd4c
> 2. Change the benchmark circuit of [fpga_verilog/bram/dpram1k](https://github.com/lnis-uofu/OpenFPGA/blob/master/openfpga_flow/tasks/fpga_verilog/bram/dpram1k/config/task.conf) task as [FIFO](https://github.com/lnis-uofu/OpenFPGA/blob/master/openfpga_flow/benchmarks/micro_benchmark/fifo/rtl/fifo.v) circuit.
> 3. Run the task.
> 4. See error.

> **Expected behavior**
> I want to benchmark a FIFO circuit on a FPGA architecture with block RAM.

> **Screenshots**
> Error logs:
> [image]
> VPR logs:
> [image]

> **Enviornment (please complete the following information):**

>  - OS: 
> - [X] Ubuntu 20.04

>  - Compiler:
> - [X] gcc-9

> - Version:
> - [X] Current master

> **Additional context**
> Add any other context about the problem here.

tangxifan, 2022-07-26:
@yunuseryilmaz18 I think this is more like a Yosys script issue. You see $alu and $DFFE_PP1P due to some mapping failures in Yosys.

Can you share which Yosys scripts you are using?

yunuseryilmaz18, 2022-07-26:
@tangxifan I'm using the [template script](https://github.com/lnis-uofu/OpenFPGA/blob/master/openfpga_flow/misc/ys_tmpl_yosys_vpr_bram_flow.ys) in the repo.

msaideroglu, 2022-09-21:
Hi @tangxifan. I'm facing same issue. How should I change Yosys script?