lnis-uofu/OpenFPGA #734: run-task compilation_verification command on docker container isn't working

z-a-p-k-i-n-g, 2022-07-28:
Hi there,

I am on Ubuntu 22.04. Following the steps on doc, I am ending up with Permission denied errors.

I tried copying the docker directory to local machine and running the command from there. But still there is some error in loading libreadpine.so.7 shared library when yosys is trying to run, which is not available for Ubuntu 22.04.

I even tried compiling from source which is resulting in even more strange errors.

I am completely clueless...any suggestions would be of great help

z-a-p-k-i-n-g, 2022-07-29:
Found a workaround (probably this is not the way it usually works for all).....

Using docker cp command I copied the /opt/openfpga/ directory in the docker container to a working directory in my local machine.

 Changing read/write permissions and
then mounting this working directory onto /opt/openfpga/ while using docker run -v command, solved the "permission denied" errors.

closed 2022-07-29 by z-a-p-k-i-n-g

ganeshgore, 2022-07-29:
Hello @z-a-p-k-i-n-g 

Thank you for your interest in the tool.
The following instructions did not work for you.

```
# To get the docker image from the repository, 
docker pull ghcr.io/lnis-uofu/openfpga-master:latest

# To invoke openfpga_shell
docker run -it ghcr.io/lnis-uofu/openfpga-master:latest openfpga/openfpga bash
```

z-a-p-k-i-n-g, 2022-07-29:
Nope they didn't work for me. It threw "command not found" error