lnis-uofu/OpenFPGA #748: Error while compiling: size of array 'altStackMem' is not an integral constant-expression

jaysunl, 2022-08-20:
I am trying to install and compile OpenFPGA using the tutorial and got this error while running ```make all -j```:
[image]
How to fix? Thanks

tangxifan, 2022-08-30:
Hello, please provide your OS and compiler information. You can follow the issue template: https://github.com/lnis-uofu/OpenFPGA/blob/master/.github/ISSUE_TEMPLATE/bug_report.md

LvmhZhD, 2022-09-16:
> 我正在尝试使用教程安装和编译 OpenFPGA 并在运行时出现此错误 `make all -j`： [image]怎么修？ 谢谢
Have you fixed this problem? I had the same problem，my ubuntu is 2204. thx

jaysunl, 2022-09-18:
@tangxifan 
I am compiling using the below OS and compiler. I tried replacing SIGSTKSZ with the constant 32768 but I then got further errors. I think I am approaching the installation process incorrectly.

> **Enviornment (please complete the following information):**

>  - OS: Ubuntu 22.04

>  - Compiler: gcc-9

> - Version: Current master

@LvmhZhD I have not managed to make progress for this error.

LvmhZhD, 2022-09-20:
> @tangxifan 我正在使用以下操作系统和编译器进行编译。我尝试用常量32768替换SIGSTKSZ，但随后我遇到了进一步的错误。我认为我错误地处理了安装过程。
> 
> > **环境（请填写以下信息）：**
> 
> > * 操作系统： 乌班图 22.04
> 
> > * 编译器： gcc-9
> 
> > * 版本：当前主站
> 
> @LvmhZhD我没有设法为这个错误取得进展。

I found a soultion :change some tool version, but errors come again subsequently .I am also 2204, maybe OpenFPGA need to run in Ubuntu 20 04 without any changing default setting.

tangxifan, 2022-09-23:
@jaysunl @LvmhZhD Hi All, please be aware that 

- We ensure OpenFPGA has no issues when compiling on Ubuntu 18.04 through CIs. 
- We have also tested on local machines with Ubuntu 20.04 and it is working fine.
- We have **not** tested on Ubuntu 22.04. There could be some issues that we currently do not have any solution.

We do plan to extend our OS support, e.g., Windows and latest ubuntu. Please give a try on latest master, where we have upgraded VtR to its latest version. See if this problem remains or not.

tangxifan, 2022-09-26:
@jaysunl @LvmhZhD The compilation errors should be gone if you use the latest OpenFPGA v1.2.x 

Close it now. If you still see issues, please let us know,

closed 2022-09-26 by tangxifan

LvmhZhD, 2022-09-27:
Works well now.Thank you