lnis-uofu/OpenFPGA #749: The reset signal is not detected on FPGA

yunuseryilmaz18, 2022-08-23:
> **Describe the bug**
> I've created a task with k4_frac_N4_tileable_fracff_40nm FPGA architecture and run the autocheck testbench on Modelsim. However, the reset signal does not affect the counter which is my benchmark circuit. It does not matter which state is in, the counter on the FPGA never resets.

> **Which part of OpenFPGA is buggy**
> [X] VPR
> [X] FPGA-Verilog

> **To Reproduce**
> Steps to reproduce the behavior:
> 1. Clone OpenFPGA repository and checkout commit id: 2a43843bd7a7e73f2fdfd9563d34cdb3016d4051
> 2. Create `fracff-test` task with `create-task` command.
> 3. Copy the `k4_frac_N4_tileable_fracff_40nm` architecture file and its annotation file to the task folder.
> 4. Copy the counter designs in the `micro-benchmark` folder to the task folder.
> 5. Execute OpenFPGA task.
> 6. See error

> **Expected behavior**
> I want to benchmark a counter circuit on a FPGA architecture with fracturable flip-flops.

> **Screenshots**
>  [image]

> **Environment (please complete the following information):**

>  - OS: 
> - [X] Ubuntu 20.04

> - Compiler:
> - [X] gcc-9

> - Version:
> - [X] Current master

> **Additional context**
> I've uploaded the task folder and my final run below:
[fracff-test.zip](https://github.com/lnis-uofu/OpenFPGA/files/9394790/fracff-test.zip)

tangxifan, 2022-08-30:
@yunuseryilmaz18 Please check the usage of 

- repack pin constraints https://openfpga.readthedocs.io/en/master/manual/openfpga_shell/openfpga_commands/fpga_bitstream_commands/#cmdoption-design_constraints
- pin constraint file https://openfpga.readthedocs.io/en/master/manual/openfpga_shell/openfpga_commands/fpga_verilog_commands/#cmdoption-pin_constraints_file

narutozxp, 2022-09-07:
I know that openfpga provides a file to describe the pin constraints, but I don't know how openfpga calls this file. Or should I use vpr --place_file place_file_name.place for my purpose?

tangxifan, 2022-09-07:
@narutozxp Please look into this example: https://github.com/lnis-uofu/OpenFPGA/tree/master/openfpga_flow/tasks/basic_tests/constrain_pin_location/config

tangxifan, 2022-09-07:
@narutozxp Yes. This is a new feature that is added recently.

tangxifan, 2022-10-20:
Close for now. If you still have question, please reopen and discuss.

closed 2022-10-20 by tangxifan