lnis-uofu/OpenFPGA #756: How can I modify the K4 skywater130nm architecture to a K3 LUT size?

Sazadur, 2022-08-31:
Hi, I am fairly new with OpenFPGA. I have been wondering if anyone can point me in the right direction to modify the k4_frac_N8_tileable_register_scan_chain_nonLR_caravel_io_skywater130nm architecture file to reduce the LUT size and LE (K3N2)? Thanks in advance,

tangxifan, 2022-08-31:
@Sazadur Please check VTR's architecture language： https://docs.verilogtorouting.org/en/latest/arch/reference/#complex-blocks

Sazadur, 2022-08-31:
@tangxifan, thank you for your response. I will look into this.

tangxifan, 2022-09-06:
Close the issue as answered. If you need further help, please comment (reopen this issue) or create a new issue.
Thank you for the interests in OpenFPGA.

closed 2022-09-06 by tangxifan