lnis-uofu/OpenFPGA #758: The regout isn't connected to anywhere in k6_frac_N10_tileable_adder_register_chain_40nm architecture

yunuseryilmaz18, 2022-09-01:
> **Describe the bug**
> I've created a fabric with the [register_chain](https://github.com/lnis-uofu/OpenFPGA/tree/master/openfpga_flow/tasks/fpga_verilog/fabric_chain/register_chain/config) task and run the autocheck testbench. The testbench works correctly but the design did not map on the register chain in the CLBs. When I read the fabric netlist codes, I saw that the fle_regin and the fle_regout signals are not connected to FFs in the BLEs. I think the VPR architecture file of this task is incomplete.

> **To Reproduce**
> Steps to reproduce the behavior:
> 1. Clone OpenFPGA repository and checkout commit id: bf382a9be5f95dd567714d5b8b7b2c4532612764
> 2. Execute OpenFPGA task `register_chain` and run the autocheck testbench.
> 3. See error

> **Expected behavior**
> I expected to see the shift register mapped on the register chain.

> **Screenshots**
> [image]
> I've uploaded the signals of CLB_1_1 but it is valid for all CLBs.

> **Enviornment (please complete the following information):**

>  - OS: 
> - [X] Ubuntu 20.04

>  - Compiler:
> - [X] gcc-9

> - Version:
> - [X] Current master

> **Additional context**
> I've added the task and the benchmark folder below.
[register_chain.zip](https://github.com/lnis-uofu/OpenFPGA/files/9469760/register_chain.zip)
[shift_reg.zip](https://github.com/lnis-uofu/OpenFPGA/files/9469769/shift_reg.zip)

tangxifan, 2022-09-01:
@yunuseryilmaz18 I would suggest not to use the shift register mode for now, because it is a bit buggy. We are still investigating if we should have such feature in CLB or not.

yunuseryilmaz18, 2022-09-06:
Thank you. I changed the physical mode of CLBs but still, it didn't map on the shift register.
[vpr_arch.zip](https://github.com/lnis-uofu/OpenFPGA/files/9493681/vpr_arch.zip)

tangxifan, 2022-09-06:
@yunuseryilmaz18 To use shift registers, you need to first ensure that the .blif model in your shift register pb_type is supported by synthesis tool (double check your tech. lib). Second, to support shift register chains, you may need to implement a yosys plugin, as  the quicklogic's yosys plugin (open-source).