lnis-uofu/OpenFPGA #772: Creating Spypads Using XML Syntax

zliu1Charlotte, 2022-09-08:
Hi I did a walk through on "Creating Spypads Using XML Syntax".
After comment out this in the conf file:
[SCRIPT_PARAM_MIN_ROUTE_CHAN_WIDTH]
end_flow_with_test=
#vpr_fpga_verilog_formal_verification_top_netlist=

I ran this: python3 openfpga_flow/scripts/run_fpga_task.py fpga_verilog/spypad --debug --show_thread_logs

but I got errors on iverilog:
RROR (00_and2_MIN_ROUTE_CHAN_WIDTH) - iverilog_verification run failed with returncode 1
ERROR (00_and2_MIN_ROUTE_CHAN_WIDTH) - command iverilog -o compiled_and2 ./SRC/and2_include_netlists.v -s and2_autocheck_top_tb -I ./SRC
ERROR (00_and2_MIN_ROUTE_CHAN_WIDTH) - -->>error: Unable to find the root module "and2_autocheck_top_tb" in the Verilog source.
ERROR (00_and2_MIN_ROUTE_CHAN_WIDTH) - -->>1 error(s) during elaboration.
ERROR (00_and2_MIN_ROUTE_CHAN_WIDTH) - Current working directory : /home/pc-1/zliu/OpenFPGA/openfpga_flow/tasks/fpga_verilog/spypad/run010/k6_frac_N10_tileable_adder_register_scan_chain_depop50_spypad_40nm/and2/MIN_ROUTE_CHAN_WIDTH
ERROR (00_and2_MIN_ROUTE_CHAN_WIDTH) - Failed to run iverilog_verification task
ERROR (00_and2_MIN_ROUTE_CHAN_WIDTH) - Exiting . . . . . .
ERROR (00_and2_MIN_ROUTE_CHAN_WIDTH) - Failed to execute openfpga flow - 00_and2_MIN_ROUTE_CHAN_WIDTH

tangxifan, 2022-09-08:
@zliu1Charlotte If you comment out the line, you are requesting openfpga flow to run a full testbench simulation. But the openfpgashell script you are using only supports preconfigured testbench generation.

Please refer to the example script:

https://github.com/lnis-uofu/OpenFPGA/blob/master/openfpga_flow/openfpga_shell_scripts/example_script.openfpga

closed 2022-09-08 by tangxifan

zliu1Charlotte, 2022-09-08:
Thanks. So this error is normal to shown in this example correct? or is there a way to fix this?
But the vcd file is not generated. 

in the iverilog_output.txt

 iverilog -o compiled_and2 ./SRC/and2_include_netlists.v -s and2_autocheck_top_tb -I ./SRC
  error: Unable to find the root module "and2_autocheck_top_tb" in the Verilog source.
       : Perhaps ``-s and2_autocheck_top_tb'' is incorrect?
  1 error(s) during elaboration.
  1

tangxifan, 2022-09-08:
@zliu1Charlotte it is normal when you modify the scripts. If you see errors in existing regression tests (no modification), it is a valid bug.

zliu1Charlotte, 2022-09-08:
Thanks, is there a way to "source iverilog_output.txt" without the error to see the vcd file?

tangxifan, 2022-09-08:
@zliu1Charlotte I believe you should change the openfpgashell script by referring to the example:

https://github.com/lnis-uofu/OpenFPGA/blob/1073c3306ba4ec3005a1392e14e422dbbc9a565a/openfpga_flow/openfpga_shell_scripts/example_script.openfpga#L58