lnis-uofu/OpenFPGA #776: Pin Constraint Explicit Port Direction

tornupnegatives, 2022-09-09:
**Is your feature request related to a problem? Please describe.**
When applying pin constraints via PCF files, the direction (input/output) is currently inferred from the name of the top-level port (_A2F/_F2A). However, in designs which to not use this naming convention, OpenFPGA cannot detect the direction of the port.

**Describe the solution you'd like**
The pcf2place parser should read the `gpio_type` column of the pin table CSV file to determine the direction of the port.

**Describe alternatives you've considered**

**Additional context**

tangxifan, 2022-09-09:
@tornupnegatives OpenFPGA reads the post-synthesis netlist (.blif) to identify the input/output for a given port. We do not use hardcoded names to identify port direction.

tornupnegatives, 2022-09-09:
Allow me to elaborate.

The architecture I am working on uses a bidirectional GPIO interface like such:
```Verilog
inout [0:47] gfpga_pad_GPIO_OpenFPGA_PAD;
```

I would like to perform pin constraining using the following generic names:
```CSV
orientation,row,col,pin_num_in_cell,port_name,mapped_pin,GPIO_type,Associated Clock,Clock Edge
TOP,,,,gfpga_pad_GPIO_OpenFPGA_PAD[0],io_i[0],,,
TOP,,,,gfpga_pad_GPIO_OpenFPGA_PAD[1],io_i[1],,,
TOP,,,,gfpga_pad_GPIO_OpenFPGA_PAD[2],io_i[2],,,
TOP,,,,gfpga_pad_GPIO_OpenFPGA_PAD[3],io_i[3],,,
TOP,,,,gfpga_pad_GPIO_OpenFPGA_PAD[4],io_i[4],,,
TOP,,,,gfpga_pad_GPIO_OpenFPGA_PAD[5],io_i[5],,,
...
```

However, OpenFPGA crashes with the following error:
```shell
Read the design constraints from a pcf file: $HOME/OpenFPGA/openfpga_flow/tasks/basic_tests/constrain_pin_location/config/and2.pcf.
Read the blif from a file: and2.blif.
Read I/O Location Map
Read I/O Location Map took 0.00 seconds (max_rss 16.7 MiB, delta_rss +0.0 MiB)
Read the I/O location map from an XML file: $HOME/mantis/flow/pin_constraint/io_map.xml.
Read I/O Pin Table
Invalid direction defintion! Expect [A2F|F2A] in the pin name
```

According to the [documentation](https://openfpga.readthedocs.io/en/master/manual/file_formats/pin_table_file/?highlight=A2F#pin-table-file-csv), pin table CSV files require the _A2F or _F2A post-fix to determine port direction:

_"Currently, the direction of the port is inferred by the port_name. A postfix of A2F indicates an input port, while a postfix of F2A indicates an output port."_

I am seeking to change this behavior such that the port direction is determined instead by the `GPIO_type` column of the pin table. Hopefully this clarifies my concern.

tangxifan, 2022-09-09:
@tornupnegatives I just double checked the codes. I believe the documentation is out-of-date. The restriction is no longer there. Thank you for the clarification.

tangxifan, 2022-09-09:
@tornupnegatives I found where the limitation comes from. Thank you for pointing this out. Will fix it.

closed 2022-10-18 by tangxifan