lnis-uofu/OpenFPGA #778: Technology Map for Hard Adder

yunuseryilmaz18, 2022-09-12:
Hi, do I need to use a technology map file for the architectures with hard adders (e.g. k6_frac_N10_adder_chain_40nm)? I saw some tasks in the repo but the circuits with adders are not used in them.

tangxifan, 2022-09-12:
@yunuseryilmaz18 You are right. Currently, we are still working on an open-source technology library and synthesis support for hard adders.

closed 2022-09-12 by tangxifan

yunuseryilmaz18, 2022-09-14:
I wrote a blif file to map on hard adders. The circuit is placed successfully, but the circuit cannot be routed in VPR flow. The vpr_stdout.log error screenshot is added below:
[image]

Also, I've added my files below:
[adder_chain.zip](https://github.com/lnis-uofu/OpenFPGA/files/9565477/adder_chain.zip)

What should I do to map on hard adders?

tangxifan, 2022-09-14:
@yunuseryilmaz18 If it is not routable, please check if ``cin`` or ``cout`` of your ``pb_type`` is accessible by global routing or not. In VPR examples, the first adder in the chain does not get input from ``cin`` (it is set to unconnected). You may need to double check your netlists.

yunuseryilmaz18, 2022-09-16:
Yes, some `cin` and `cout` pins are not connected to the global routing architecture but I want to implement an adder carry chain on a CLB. What should I do?