lnis-uofu/OpenFPGA #788: Full testbench is working but preconfigured testbench and behavioral benchmark are giving 1'bX

msaideroglu, 2022-09-15:
I um running /openfpga_flow/tasks/fpga_verilog/bram/dpram1k task. It successfully completes as displayed in shell. However preconfigured testbench can not display true results. All Ram input and outputs are displaying 1'bx as well as behavioral benchmark ports. However if I run full_testbench, benchmark ports displays 1'bx again but fpga ports display logic values, but I can not check if the design is correctly placed and routed. When I rearrange autochecked_testbench to do a proper test on dpram, I can see that Ram is correctly working however still have same problem with benchmark results and preconfigured testbench results. I also tried to comment out LINE 58 full_testbench line to get true results for preconfigured testbench, result was same as before. I attached preconf and full testbench views:
[image]
(preconfigured tb results with 1'bx's)
[image]
(full tb results with wrong benchmark IOs)

To reproduce:
run "run-task dpram1k" in the task folder

tangxifan, 2022-09-15:
@msaideroglu I think it is due to the lack of initial states of the RAM block. 

https://github.com/lnis-uofu/OpenFPGA/blob/b0b3d52e6680de15e4cd53b69eec7ea30e3a5cd4/openfpga_flow/openfpga_yosys_techlib/openfpga_brams_sim.v#L22-L49

Can you try to add initial states like: 

```
// Initial values are all random, to mimic the actual behavoir of a RAM
initial begin
  for (i = 0; i < DEPTH; i = i + 1) begin
    ram[i] = $random;
  end
  q_reg <= $random;
end
```

msaideroglu, 2022-09-19:
Hi @tangxifan. I added that initial state in **_OpenFPGA/openfpga_flow/openfpga_yosys_techlib/k6_frac_N10_tileable_adder_chain_mem1K_40nm_cell_sim.v_** . However the result was same as before.

tangxifan, 2022-09-23:
@msaideroglu Sorry. The initial states should be added to your HDL design rather than the ``cell_sim.v``

tangxifan, 2022-10-20:
@msaideroglu Let me know if the issue still exist. Close the issue for now.

closed 2022-10-20 by tangxifan