lnis-uofu/OpenFPGA #790: Add direction information in GSB output of CB

ganeshgore, 2022-09-15:
**Is your feature request related to a problem? Please describe.**
Currently, the GSB information extracted for the connection box from the OpenFPGA has the following structure.
The side attribute for driver_node always shows `left` for `CHANX` and `top` for `CHANY`, Whereas these channels are coming from two different sides, it will be good to have that information captured in the side parameters.
Current alternative uses  the index number to estimate the direction, as the general channel structure always seems to alternate between two directions.
```
 <IPIN side="top" index="0" node_id="202" mux_size="8">
    <driver_node type="CHANX" side="left" node_id="1520" index="0" segment_id="0"/>
    <driver_node type="CHANX" side="left" node_id="1521" index="1" segment_id="0"/>
    <driver_node type="CHANX" side="left" node_id="1532" index="12" segment_id="0"/>
....
 </IPIN>
```

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

tangxifan, 2022-09-15:
@ganeshgore Currently, direct connections between tiles, as defined in the ``<directlist>`` of VPR's architecture description file, are not considered as part of the GSB. They are implemented/instantiated at the top-level module. This helps us to reduce the number of unique modules. We can output another file containing these information for you, if needed.

tangxifan, 2022-09-15:
O.K. You want to know the direction of these uni-directional routing tracks? This can be quickly done.