lnis-uofu/OpenFPGA #791: Add a benchmark in the softadder example

zliu1Charlotte, 2022-09-15:
I want to adopt the softadder example from OpenFPGA/openfpga_flow/tasks/quicklogic_tests/lut_adder_test
to a 8 bit counter follow up with a Sbox (my own example). 

in the task.conf
i added:  bench4=${PATH:OPENFPGA_PATH}/openfpga_flow/benchmarks/micro_benchmark/adder/counterSBox/*.v
and 
added: bench4_top = counterSBox

these remains the same:
[OpenFPGA_SHELL]
openfpga_shell_template=${PATH:OPENFPGA_PATH}/openfpga_flow/openfpga_shell_scripts/bitstream_setting_example_script.openfpga
openfpga_arch_file=${PATH:OPENFPGA_PATH}/openfpga_flow/openfpga_arch/k4_frac_N8_reset_softadderSuperLUT_register_scan_chain_caravel_io_skywater130nm_fdhd_cc_openfpga.xml
openfpga_sim_setting_file=${PATH:OPENFPGA_PATH}/openfpga_flow/openfpga_simulation_settings/fixed_sim_openfpga.xml
openfpga_bitstream_setting_file=${PATH:OPENFPGA_PATH}/openfpga_flow/tasks/quicklogic_tests/lut_adder_test/config/bitstream_annotation.xml
openfpga_vpr_circuit_format=eblif
 
 [ARCHITECTURES]
 arch0=${PATH:OPENFPGA_PATH}/openfpga_flow/vpr_arch/k4_frac_N8_tileable_reset_softadderSuperLUT_register_scan_chain_nonLR_caravel_io_skywater130nm.xml

But I got error:
(00_counterSBox_MIN_ROUTE_CHAN_WIDTH) - Run directory : /home/pc-1/zliu/OpenFPGA/openfpga_flow/tasks/quicklogic_tests/lut_adder_test/run007/k4_frac_N8_tileable_reset_softadderSuperLUT_register_scan_chain_nonLR_caravel_io_skywater130nm/counterSBox/MIN_ROUTE_CHAN_WIDTH
 INFO (00_counterSBox_MIN_ROUTE_CHAN_WIDTH) - Running "yosys_vpr" Flow
 INFO (00_counterSBox_MIN_ROUTE_CHAN_WIDTH) - Extracted lut_size size from arch XML = 4
 INFO (00_counterSBox_MIN_ROUTE_CHAN_WIDTH) - Running Yosys with lut_size = 4
 INFO (00_counterSBox_MIN_ROUTE_CHAN_WIDTH) - Launching Run yosys 
 INFO (00_counterSBox_MIN_ROUTE_CHAN_WIDTH) - Run yosys is written in file yosys_output.log
 INFO (00_counterSBox_MIN_ROUTE_CHAN_WIDTH) - Extracted lut_size size from arch XML = 4
 INFO (00_counterSBox_MIN_ROUTE_CHAN_WIDTH) - Running Yosys with lut_size = 4
 INFO (00_counterSBox_MIN_ROUTE_CHAN_WIDTH) - Yosys rewrite iteration: 0
 INFO (00_counterSBox_MIN_ROUTE_CHAN_WIDTH) - Launching Run yosys 
 INFO (00_counterSBox_MIN_ROUTE_CHAN_WIDTH) - Run yosys is written in file yosys_rewrite_output_0.log
 INFO (00_counterSBox_MIN_ROUTE_CHAN_WIDTH) - Yosys rewrite iteration: 1
 INFO (00_counterSBox_MIN_ROUTE_CHAN_WIDTH) - Launching Run yosys 
 INFO (00_counterSBox_MIN_ROUTE_CHAN_WIDTH) - Run yosys is written in file yosys_rewrite_output_1.log
 INFO (00_counterSBox_MIN_ROUTE_CHAN_WIDTH) - Running OpenFPGA Shell Engine 
 INFO (00_counterSBox_MIN_ROUTE_CHAN_WIDTH) - Launching OpenFPGA Shell Run 
 INFO (00_counterSBox_MIN_ROUTE_CHAN_WIDTH) - OpenFPGAShell Revision: 164a788c-dirty Compiled: 2022-09-06T21:48:56
ERROR (00_counterSBox_MIN_ROUTE_CHAN_WIDTH) - OpenFPGA Shell Run run failed with returncode 1
ERROR (00_counterSBox_MIN_ROUTE_CHAN_WIDTH) - command /home/pc-1/zliu/OpenFPGA/openfpga/openfpga -batch -f counterSBox_run.openfpga
ERROR (00_counterSBox_MIN_ROUTE_CHAN_WIDTH) - -->>Error 1: 
ERROR (00_counterSBox_MIN_ROUTE_CHAN_WIDTH) - -->>Error 2: Command 'vpr' execution has fatal errors
ERROR (00_counterSBox_MIN_ROUTE_CHAN_WIDTH) - Current working directory : /home/pc-1/zliu/OpenFPGA/openfpga_flow/tasks/quicklogic_tests/lut_adder_test/run007/k4_frac_N8_tileable_reset_softadderSuperLUT_register_scan_chain_nonLR_caravel_io_skywater130nm/counterSBox/MIN_ROUTE_CHAN_WIDTH
ERROR (00_counterSBox_MIN_ROUTE_CHAN_WIDTH) - Failed to run OpenFPGA Shell Run task
ERROR (00_counterSBox_MIN_ROUTE_CHAN_WIDTH) - Exiting . . . . . .
ERROR (00_counterSBox_MIN_ROUTE_CHAN_WIDTH) - Failed to execute openfpga flow - 00_counterSBox_MIN_ROUTE_CHAN_WIDTH

Is this not the way to add another benchmark?

tangxifan, 2022-09-15:
@zliu1Charlotte If you want to use the SuperLUT feature, the synthesis recipe is different. Please check the test case in quicklogic_test

zliu1Charlotte, 2022-09-15:
is there a good example that contains with a soft design and skywater tech cell?

tangxifan, 2022-09-15:
@zliu1Charlotte If you cannot find it the SOFA repo, then there is none.

zliu1Charlotte, 2022-09-15:
Thank you for using OpenFPGA!
 Error 1:
 Type: Routing
 File: /home/pc-1/zliu/OpenFPGA/vpr/src/base/place_and_route.cpp
 Line: 165
 Message: This circuit requires a channel width above 1000, probably is not going to route.
 Aborting routing procedure.
 
 Error 2: Command 'vpr' execution has fatal errors
 1

I got an error like this. Does this mean the i need a bigger fpga arch?

closed 2022-09-16 by zliu1Charlotte