lnis-uofu/OpenFPGA #793: This circuit requires a channel width above 1000, probably is not going to route.

zliu1Charlotte, 2022-09-16:
i am doing a custom design but it failed on routing. 
I am using soft_adder example

openfpga_shell_template=${PATH:OPENFPGA_PATH}/openfpga_flow/openfpga_shell_scripts/bitstream_setting_example_script.openfpga
  openfpga_arch_file=${PATH:OPENFPGA_PATH}/openfpga_flow/openfpga_arch/k4_frac_N8_reset_softadder_register_scan_chain_caravel_io_skywater130nm_fdhd_cc_openfpga.xml
  openfpga_sim_setting_file=${PATH:OPENFPGA_PATH}/openfpga_flow/openfpga_simulation_settings/fixed_sim_openfpga.xml
  openfpga_bitstream_setting_file=${PATH:OPENFPGA_PATH}/openfpga_flow/tasks/fpga_verilog/adder/soft_adder/config/bitstream_annotation.xml
  openfpga_vpr_circuit_format=eblif
  external_fabric_key_file=

Have you seen this error before?

 (00_counterSBox_MIN_ROUTE_CHAN_WIDTH) - OpenFPGA Shell Run run failed with returncode 1
ERROR (00_counterSBox_MIN_ROUTE_CHAN_WIDTH) - command /home/pc-1/zliu/OpenFPGA/openfpga/openfpga -batch -f counterSBox_run.openfpga
ERROR (00_counterSBox_MIN_ROUTE_CHAN_WIDTH) - -->>Error 1: 
ERROR (00_counterSBox_MIN_ROUTE_CHAN_WIDTH) - -->>Error 2: Command 'vpr' execution has fatal errors
ERROR (00_counterSBox_MIN_ROUTE_CHAN_WIDTH) - Current working directory : /home/pc-1/zliu/OpenFPGA/openfpga_flow/tasks/fpga_verilog/adder/soft_adder/run010/k4_frac_N8_tileable_reset_softadder_register_scan_chain_nonLR_caravel_io_skywater130nm/counterSBox/MIN_ROUTE_CHAN_WIDTH
ERROR (00_counterSBox_MIN_ROUTE_CHAN_WIDTH) - Failed to run OpenFPGA Shell Run task
ERROR (00_counterSBox_MIN_ROUTE_CHAN_WIDTH) - Exiting . . . . . .
ERROR (00_counterSBox_MIN_ROUTE_CHAN_WIDTH) - Failed to execute openfpga flow - 00_counterSBox_MIN_ROUTE_CHAN_WIDTH

closed 2022-09-18 by zliu1Charlotte

msaideroglu, 2022-09-19:
Hi @zliu1Charlotte . How did you solve this issue? I'm facing the same error.

zliu1Charlotte, 2022-09-19:
this did not resolved, i went with other task.conf file sorry.